platformio/platformio-core #4440: Invalid escape character in c_cpp_properties.json

brunojoyal, 2022-10-17:
VSCode is suddenly unhappy with c_cpp_properties.json. My platformio.ini file contains the build flag

`    -DTEMPLATE_PLACEHOLDER="\'\~\'"`

which yields to the following in c_cpp_properties.json:

[image]

with the message "Invalid escape character in string. jsonc(261)"

Everything compiles just fine but I have to manually fix c_cpp_properties.json in order for Intellisense to work properly. I did not have this problem before but I'm not sure what caused it. Any help appreciated!

elmout31, 2022-11-24:
Same since upgrade to version 1.73.1.  If i remove the escape chars, Intellisense, it works again. Nevertheless, vscode automatically set them back and so at each vscode restart i have to repeat the operation.

ivankravets, 2022-11-24:
What is your `platformio.ini`?

ivankravets, 2023-04-21:
What should be the real value of `TEMPLATE_PLACEHOLDER`?

jlaustill, 2024-08-04:
I'm this same issue with a different build flag. What is adding these extra slashes? It is driving me nuts!