platformio/platformio-core #4441: Warning! Ignore unknown configuration option `monitor_flags` in section [env]

XeroZar, 2022-10-22:
Has anyone found a solution for this warning... I've looked at the documentation 

https://docs.platformio.org/en/latest/projectconf/section_env_monitor.html#monitor-flags

and it doesn't help...

I am trying to fix this, if anyone know how, I can't figure it out if my life depended on it.

==Code Below==

monitor_flags =
  --quiet
  --echo
  --eol
    LF
  --filter
    colorize
  --filter
    time

==End Code==

maxgerhardt, 2022-10-26:
```ini
monitor_echo = yes
monitor_eol = LF
monitor_filters =
   colorize
   time
```

ivankravets, 2022-10-26:
Thanks, @maxgerhardt, for the solution.

closed 2022-10-26 by ivankravets

XeroZar, 2022-10-26:
> ```ini
> monitor_echo = yes
> monitor_eol = LF
> monitor_filters =
>    colorize
>    time
> ```

Your awesome thank you.

N0YHR, 2023-11-11:
I know this is closed, but I've searched numerous files within the project hoping to find "monitor_echo", but do not see where these edits go to eliminate the error.

N0YHR, 2023-11-11:
I just found them in platformio.ini

Thanks...

sancio1234, 2024-05-25:
dove e cosa hai fato

sancio1234, 2024-05-25:
> Li ho appena trovati in platformio.ini
> 
> Grazie...

come lo ai compilato che non riesco

sancio1234, 2024-05-25:
> > ```ini
> > monitor_echo = yes
> > monitor_eol = LF
> > monitor_filters =
> >    colorize
> >    time
> > ```
> 
> Il tuo fantastico grazie.

dove lo devo copiare in piu mi dice no section headers