platformio/platformio-core #4446: pio check using clangtidy fails on large project

grisharav, 2022-10-31:
What kind of issue is this?

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 11 21H2

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.5rc1
Tested on latest available version 

### Description of problem
When running `pio check` I get an exception:
```
Checking gpt_pcb > cppcheck (board: pico32; framework: arduino; platform: espressif32)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
No defects found
=========================================================================== [PASSED] Took 76.57 seconds ===========================================================================
Checking gpt_pcb > clangtidy (board: pico32; framework: arduino; platform: espressif32)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-clangtidy @ ~1.120001.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-clangtidy@1.120001.0 has been installed!
Error: Traceback (most recent call last):
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\platformio\cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\platformio\check\cli.py", line 136, in cli
    rc = ct.check(
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\platformio\check\tools\base.py", line 236, in check
    self.execute_check_cmd(cmd)
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\platformio\check\tools\base.py", line 185, in execute_check_cmd
    result = proc.exec_command(
  File "C:\Users\GregRavikovich\.platformio\penv\lib\site-packages\platformio\proc.py", line 114, in exec_command
    with subprocess.Popen(*args, **kwargs) as p:
  File "C:\Users\GregRavikovich\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\GregRavikovich\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 206] The filename or extension is too long

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```
This started to happen after the project has grown enough.

I do have long path enabled, see the discussion under https://community.platformio.org/t/filename-too-long-when-running-check/28986
#### Steps to Reproduce

1. Create a large project (unfortunately can't submit it, proprietary)
2. Setup check using clangtidy
3. run `pio check`

### Actual Results

The error above

### Expected Results

Check results available.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env]
build_flags = 
	!python ./git_rev_macro/git_rev_macro.py
	-std=gnu++17
build_unflags =
    -std=gnu++11
board = pico32
framework = arduino
platform = espressif32@5.1.1
monitor_speed = 115200
build_type = release
check_tool = cppcheck, clangtidy
check_skip_packages = yes
check_patterns = lib/sgtcm_lib/*
check_flags = 
	cppcheck: --suppress=*:*.pio\* --suppress=noExplicitConstructor --suppress=unusedFunction --suppress=*:*lib\sgtcm_lib\src\SolarPosition.cpp
	clangtidy: --checks=clang-analyzer-*,readability-*,performance-*,hicpp-*,bugprone-*,modernize-*,-modernize-use-trailing-return-type --suppress=readability-qualified-auto
monitor_filters = esp32_exception_decoder
lib_ldf_mode = deep
custom_nanopb_protos = 
	+<../../shared_models/tracker_models.proto>
custom_nanopb_options =
    --error-on-unmatched --cpp-descriptors
lib_deps =
	rlogiacco/CircularBuffer@^1.3.3
	lasselukkari/aWOT@^3.4.0
	paulstoffregen/Time@^1.6.1
	hideakitai/ArduinoEigen@^0.2.3
	nanopb/Nanopb@0.4.6+3

[env:gpt_pcb]
build_flags = ${env.build_flags} -D GRID_POWER_TRACKER
debug_tool = esp-prog
debug_init_break = tbreak setup

lib_deps =
	${env.lib_deps}
	sparkfun/SparkFun L6470 AutoDriver@^1.3.3

[env:spt_pcb]
build_flags = ${env.build_flags} -D SELF_POWER_TRACKER

debug_init_break = tbreak setup
debug_tool = esp-prog

lib_deps =
	${env.lib_deps}
    teemuatlut/TMC2130Stepper@^2.5.1
    gin66/FastAccelStepper@^0.27.5
    felias-fogg/SlowSoftI2CMaster@^1.0.0
```
Note that removing all flags from `clangtidy` doesn't change the error

**Source file to reproduce issue:**

Sorry, can't provide full source

### Additional info

grisharav, 2022-11-01:
I think it's a matter of command line string length. See https://stackoverflow.com/questions/3205027/maximum-length-of-command-line-string for example. Should consider using some sort of a response file instead of direct command line invocation.
Here's the output when running with the `--verbose` flag:
```
C:\Users\GregRavikovich\.platformio\packages\tool-clangtidy\clang-tidy --quiet --checks=clang-analyzer-*,readability-*,performance-*,hicpp-*,bugprone-*,modernize-*,-modernize-use-trailing-return-type --suppress=readability-qualified-auto C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\operation_context.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\sg_communicator.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\sg_controller.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\sg_controller_factory.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\sg_handlers.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\sg_parsing.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\sg_time_format.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\esp32\esp32_controller_factory.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\esp32\esp32_pinout.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\esp32\esp32_wifi_network_manager.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\esp32\sg_esp32_platform.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\handlers\motion_status_handler.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\handlers\sg_engineering_command_handler.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\handlers\sg_mcs_command_handler.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\handlers\sg_metadata_handler.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\handlers\sg_set_smart_tracking_points_command_handler.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\handlers\sg_set_state_command_handler.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\handlers\sg_status_handler.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\handlers\sg_update_ota_command_handler.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\health\keep_alive_manager.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\health\safety_monitor.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\homing\homing_accuracy.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\homing\homing_magnet_manager.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\homing\homing_procedure_type.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\homing\homing_sensor_manager.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\homing\homing_target.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\logger\log_manager.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\logger\log_message_params_writers.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\logger\log_message_writer.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\logger\log_record.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\logger\log_severity.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\motor\motor_controller_base.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\motor\motor_specific_params_defines.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\motor\l6470\l6470_motor_controller.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\motor\l6470\l6470_motor_driver_status.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\motor\tmc2130\tmc2130_motor_controller.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\motor\tmc2130\tmc2130_motor_driver_status.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\movement\basic_movement_owner.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\movement\homing_movement_owner.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\movement\tracking\analytical_backtracking_calculator.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\movement\tracking\smart_tracking_calculator.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\movement\tracking\tracking_calculator.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\movement\tracking\tracking_movement_owner.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\network\network_connection_params.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\network\wifi_network_manager_base.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\orientation\orientation_data.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\orientation\orientation_manager.cpp C:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_lib\src\params\params_base.c [... cut ...] ol\SGTCM-PlatformIo\lib\sgtcm_simulation\src -IC:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_test_lib\include -IC:\workspace\control\local-control\SGTCM-PlatformIo\lib\sgtcm_test_lib\src -IC:\workspace\control\local-control\SGTCM-PlatformIo\.pio\libdeps\gpt_pcb\Unity\src -IC:\workspace\control\local-control\SGTCM-PlatformIo\.pio\libdeps\gpt_pcb\fmt\include -IC:\workspace\control\local-control\SGTCM-PlatformIo\.pio\libdeps\gpt_pcb\fmt\src
```
which is more than 34.2K characters...

Abasz, 2023-03-09:
clang-tidy can take a yaml file (`--config-file=.clang-tidy`) that could include the necessary include path in the ExtraArgs property. This way these do not need to be passed as cmd argument and the command can be shortened.

Bascy, 2023-03-31:
Would love to see a fix for this

nschurando, 2023-04-18:
Same here. Using the `--verbose` argument, I was able to confirm that platformio generates for my project a 39376 character long command line call to clang-tidy.

PBrunot, 2023-04-23:
It fails even with a simple project, provided the toolchain / libraries have enough included files.

Example with --verbose:

```cmd
C:\Users\pasca\.platformio\packages\tool-clangtidy\clang-tidy --quiet --checks=-*,cert-*,clang-analyzer-* --fix C:\Users\pasca\Git\rfid-arduino\src\AuthProvider.cpp C:\Users\pasca\Git\rfid-arduino\src\BoardState.cpp C:\Users\pasca\Git\rfid-arduino\src\FabServer.cpp C:\Users\pasca\Git\rfid-arduino\src\Machine.cpp C:\Users\pasca\Git\rfid-arduino\src\main.cpp C:\Users\pasca\Git\rfid-arduino\src\RFIDWrapper.cpp C:\Users\pasca\Git\rfid-arduino\include\AuthProvider.h C:\Users\pasca\Git\rfid-arduino\include\Board.h C:\Users\pasca\Git\rfid-arduino\include\BoardState.h C:\Users\pasca\Git\rfid-arduino\include\card.h C:\Users\pasca\Git\rfid-arduino\include\conf.h C:\Users\pasca\Git\rfid-arduino\include\FabServer.h C:\Users\pasca\Git\rfid-arduino\include\FabUser.h C:\Users\pasca\Git\rfid-arduino\include\LCDWrapper.h C:\Users\pasca\Git\rfid-arduino\include\Machine.h C:\Users\pasca\Git\rfid-arduino\include\pins.h C:\Users\pasca\Git\rfid-arduino\include\RFIDWrapper.h C:\Users\pasca\Git\rfid-arduino\include\secrets.h -- -DPLATFORMIO=60106 -DARDUINO_ESP32S3_DEV -DARDUINO_USB_MODE=1 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DPINS_ESP32S3 -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER="v4.4.4" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -DARDUINO_ARCH_ESP32 -DESP32 -DF_CPU=160000000L -DARDUINO=10812 -DARDUINO_VARIANT="esp32s3" -DARDUINO_BOARD="Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)" -DARDUINO_PARTITION_default -D__DBL_MIN_EXP__=(-1021) -D__FLT32X_MAX_EXP__=1024 -D__cpp_attributes=200809 -D__cpp_nontype_template_parameter_auto=201606 -D__UINT_LEAST16_MAX__=0xffff -D__ATOMIC_ACQUIRE=2 -D__FLT_MIN__=1.1754943508222875e-38F -D__GCC_IEC_559_COMPLEX=0 -D__cpp_aggregate_nsdmi=201304 -D__UINT_LEAST8_TYPE__=unsigned char -D__INTMAX_C(c)=c ## LL -D__CHAR_BIT__=8 -D__UINT8_MAX__=0xff -D__WINT_MAX__=0xffffffffU -D__FLT32_MIN_EXP__=(-125) -D__cpp_static_assert=201411 -D__ORDER_LITTLE_ENDIAN__=1234 -D__SIZE_MAX__=0xffffffffU -D__WCHAR_MAX__=0xffff -D__GCC_HAVE_SYNC_COMPARE_AND_SWAP_1=1 -D__GCC_HAVE_SYNC_COMPARE_AND_SWAP_2=1 -D__GCC_HAVE_SYNC_COMPARE_AND_SWAP_4=1 -D__DBL_DENORM_MIN__=double(4.9406564584124654e-324L) -D__GCC_ATOMIC_CHAR_LOCK_FREE=2 -D__GCC_IEC_559=0 -D__FLT32X_DECIMAL_DIG__=17 -D__FLT_EVAL_METHOD__=0 -D__cpp_enumerator_attributes=201411 -D__cpp_binary_literals=201304 -D__FLT64_DECIMAL_DIG__=17 -D__cpp_noexcept_function_type=201510 -D__GCC_ATOMIC_CHAR32_T_LOCK_FREE=2 -D__cpp_variadic_templates=200704 -D__UINT_FAST64_MAX__=0xffffffffffffffffULL -D__SIG_ATOMIC_TYPE__=int -D__DBL_MIN_10_EXP__=(-307) -D__FINITE_MATH_ONLY__=0 -D__cpp_variable_templates=201304 -D__GNUC_PATCHLEVEL__=0 -D__FLT32_HAS_DENORM__=1 -D__UINT_FAST8_MAX__=0xffffffffU -D__has_include(STR)=__has_include__(STR) -D__FLT32_MAX_10_EXP__=38 -D__DEC64_MAX_EXP__=385 -D__INT8_C(c)=c -D__INT_LEAST8_WIDTH__=8 -D__cpp_variadic_using=201611 -D__UINT_LEAST64_MAX__=0xffffffffffffffffULL -D__cpp_capture_star_this=201603 -D__SHRT_MAX__=0x7fff -D__LDBL_MAX__=1.7976931348623157e+308L -D__cpp_if_constexpr=201606 -D__UINT_LEAST8_MAX__=0xff -D__GCC_ATOMIC_BOOL_LOCK_FREE=2 -D__UINTMAX_TYPE__=long long unsigned int -D__DEC32_EPSILON__=1E-6DF -D__FLT_EVAL_METHOD_TS_18661_3__=0 -D__CHAR_UNSIGNED__=1 -D__UINT32_MAX__=0xffffffffU -D__GXX_EXPERIMENTAL_CXX0X__=1 -D__LDBL_MAX_EXP__=1024 -D__WINT_MIN__=0U -D__INT_LEAST16_WIDTH__=16 -D__SCHAR_MAX__=0x7f -D__WCHAR_MIN__=0 -D__INT64_C(c)=c ## LL -D__DBL_DIG__=15 -D__GCC_ATOMIC_POINTER_LOCK_FREE=2 -D__SIZEOF_INT__=4 -D__SIZEOF_POINTER__=4 -D__GCC_ATOMIC_CHAR16_T_LOCK_FREE=2 -D__cpp_aligned_new=201606 -D__USER_LABEL_PREFIX__ -D__STDC_HOSTED__=1 -D__LDBL_HAS_INFINITY__=1 -D__XTENSA_EL__=1 -D__FLT32_DIG__=6 -D__FLT_EPSILON__=1.1920928955078125e-7F -D__GXX_WEAK__=1 -D__SHRT_WIDTH__=16 -D__LDBL_MIN__=2.2250738585072014e-308L -D__DEC32_MAX__=9.999999E96DF -D__cpp_threadsafe_static_init=200806 -D__FLT32X_HAS_INFINITY__=1 -D__INT32_MAX__=0x7fffffff -D__INT_WIDTH__=32 -D__SIZEOF_LONG__=4 -D__UINT16_C(c)=c -D__PTRDIFF_WIDTH__=32 -D__DECIMAL_DIG__=17 -D__FLT64_EPSILON__=2.2204460492503131e-16F64 -D__INTMAX_WIDTH__=64 -D__FLT64_MIN_EXP__=(-1021) -D__has_include_next(STR)=__has_include_next__(STR) -D__LDBL_HAS_QUIET_NAN__=1 -D__FLT64_MANT_DIG__=53 -D__GNUC__=8 -D__cpp_delegating_constructors=200604 -D__FLT_HAS_DENORM__=1 -D__SIZEOF_LONG_DOUBLE__=8 -D__BIGGEST_ALIGNMENT__=16 -D__STDC_UTF_16__=1 -D__FLT64_MAX_10_EXP__=308 -D__FLT32_HAS_INFINITY__=1 -D__DBL_MAX__=double(1.7976931348623157e+308L) -D__cpp_raw_strings=200710 -D__INT_FAST32_MAX__=0x7fffffff -D__DBL_HAS_INFINITY__=1 -D__cpp_deduction_guides=201611 -D__cpp_fold_expressions=201603 -D__DEC32_MIN_EXP__=(-94) -D__INTPTR_WIDTH__=32 -D__FLT32X_HAS_DENORM__=1 -D__INT_FAST16_TYPE__=int -D__LDBL_HAS_DENORM__=1 -D__cplusplus=201703L -D__cpp_ref_qualifiers=200710 -D__DEC128_MAX__=9.999999999999999999999999999999999E6144DL -D__INT_LEAST32_MAX__=0x7fffffff -D__DEC32_MIN__=1E-95DF -D__DEPRECATED=1 -D__cpp_rvalue_references=200610 -D__DBL_MAX_EXP__=1024 -D__WCHAR_WIDTH__=16 -D__FLT32_MAX__=3.4028234663852886e+38F32 -D__DEC128_EPSILON__=1E-33DL -D__PTRDIFF_MAX__=0x7fffffff -D__FLT32_HAS_QUIET_NAN__=1 -D__GNUG__=8 -D__LONG_LONG_MAX__=0x7fffffffffffffffLL -D__SIZEOF_SIZE_T__=4 -D__cpp_rvalue_reference=200610 -D__cpp_nsdmi=200809 -D__SIZEOF_WINT_T__=4 -D__LONG_LONG_WIDTH__=64 [... cut ...]
FileNotFoundError: [WinError 206] Nome del file o estensione troppo lunga

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
``

PBrunot, 2024-02-11:
With clang-tidy ( LLVM version 17.0.6 ) I can't find a workaround.

I tried to put all the long command-line arguments into a .clang-tidy YAML file with ExtraArgs : '<contents>' and manually run clang-tidy --config-file=.clang.tidy but was not successfull.

Command line :

```ps
clang-tidy --config-file=.clang-tidy <my_cpp_file>.cpp
```

Clang.tidy file:

```yaml
# Disable all checks except for those starting with "cert-" and "clang-analyzer-"
Checks: 
  - 'cert-*'
  - 'clang-analyzer-*'

# ExtraArgs
ExtraArgs: 
  - ' -- -DPLATFORMIO=60113 -DARDUINO_ESP32S3_DEV -DARDUINO_USB_MODE=1 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DGIT_VERSION="0.3.2-2-gec4c5bc" -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER="v4.4.5" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -DARDUINO_ARCH_ESP32 -DESP32 -DF_CPU=160000000L -DARDUINO=10812 -DARDUINO_VARIANT="esp32s3" -DARDUINO_BOARD="Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)" -DARDUINO_PARTITION_default -DCORE_DEBUG_LEVEL=4 
  <omitted>
```

clang-tidy fails not finding the headers who are nevertheless present in the ExtraArgs as "-Ic:\path\to\folder" 

I tried ExtraArgsBefore instead of ExtraArgs, no success.

PBrunot, 2024-02-13:
I found a workaround : using vscode under Linux. At least this proves the issue is due to command line arguments length.