platformio/platformio-core #4447: Support Move38'

TiiFuchs, 2022-10-31:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

maxgerhardt, 2022-10-31:
Which of the 4 boards referenced in the [`boards.txt`](https://github.com/bigjosh/Move38-Arduino-Platform/blob/main/boards.txt) do you have for testing and what programmer do you use? My understanding is that this needs an ISP programmer.

maxgerhardt, 2022-10-31:
I've started a minimal project at https://github.com/maxgerhardt/pio-mov38-test that couples with repo that has the necessary platform-atmelavr modifications, it builds the firmware for a "Move38 -> Blink" board exactly like the Arduino IDE.

[image]

Uploading options not yet implemented, other boards not yet implemented.

TiiFuchs, 2022-11-01:
I have a bunch of "Blinks" with a USBtinyISP programmer to test.