platformio/platformio-core #4448: PIO warring:Can not remove temporary directory 'myproject\.pio\build`. Please remove it manually to avoid build issues

solawc, 2022-11-01:
This situation suddenly occurred to me, but other projects do not have this situation. It should be caused by the project that does not know what is set there, but I do not know how to troubleshoot this problem. So how can I know the cause of this problem? This will cause me to recompile after cleaning again, and my code will be modified correctly.

ivankravets, 2022-11-01:
Do you use antivirus tools?

solawc, 2022-11-01:
> Do you use antivirus tools?

No, my computer does not install third-party software or plug-ins except for the built-in defense of Windown11. This happens to multiple computers in this project. It can be ruled out that it is a computer problem. Need I upload the project?

Jason2866, 2022-11-18:
I did run in this error message already too. It was a not correct platformio ini setup "somewhere".

ivankravets, 2022-11-18:
@Jason2866 can you reproduce this issue again?

Jason2866, 2022-11-18:
iirc it was a wrong use of extends command in an env overriding entrys of the orig defined env.
This ended in this error or endless intellisense build.
Tbh, i dont think i can reproduce (it was a while ago), dont remember the exact config.
It was an excessive use of extends. Try to override of settings done before in an env.
The solution was not to use extends multiple times.

If @solawc  provides the platformio config which generates this error, i believe we would find the reason for.

ivankravets, 2023-04-21:
Should be fixed in the latest development branch.

closed 2023-04-21 by ivankravets