platformio/platformio-core #4454: Precompiled libraries include path doesn't include FPU variant

mike1808, 2022-11-06:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:  Linux workstation 5.15.71-1-MANJARO

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 6.1.6a`

### Description of problem

Some precompiled arduino libraries have the FPU variant in the source and include the library depending on it.
For instance, [Adafruit nRFCrypto](https://github.com/adafruit/Adafruit_nRFCrypto/tree/master/src/cortex-m4) library has `[fpv4-sp-d16-hard](https://github.com/adafruit/Adafruit_nRFCrypto/tree/master/src/cortex-m4/fpv4-sp-d16-hard)

#### Steps to Reproduce
1. Create a new project with nrf52 platform and Arduino framework and add nRFCrypto to the depedencies:
  ```ini
[env:nrf52]
platform = nordicnrf52
board = adafruit_feather_nrf52840
framework = arduino
lib_deps = 
	adafruit/Adafruit nRFCrypto@^0.0.7
  ```
2. Compile

### Actual Results

Fails to link:
```
Linking .pio/build/adafruit_feather_nrf52840/firmware.elf
/home/USER/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: cannot find -lnrf_cc310_0.9.13-no-interrupts
collect2: error: ld returned 1 exit status
*** [.pio/build/xiaonrf_adafruit/firmware.elf] Error 1
```

### Expected Results

To successfully link.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nrf52]
platform = nordicnrf52
board = adafruit_feather_nrf52840
framework = arduino
lib_deps = 
	adafruit/Adafruit nRFCrypto@^0.0.7```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

maxgerhardt, 2022-11-09:
Related: https://github.com/platformio/platformio-core/issues/4056

Offending code:

https://github.com/platformio/platformio-core/blob/b11925a9ecf0a1a2d094461095253006672de32a/platformio/builder/tools/piolib.py#L607-L617

Doesn't work with https://github.com/adafruit/Adafruit_nRFCrypto/blob/master/library.properties

maxgerhardt, 2024-01-19:
I think anothiner user just hit this in https://community.platformio.org/t/motionfx-library-compile-failed-under-the-pio-but-success-under-arduino-ide/37975/3?u=maxgerhardt: With 
```ini
[env:nucleo_f401re]
platform = ststm32
board = nucleo_f401re
framework = arduino
lib_deps = 
   stm32duino/STM32duino MotionFX @ ^1.1.0
```
it does not work out of the box either, it doesn't link in the the right library at all (no `-lMotionFX`) and from the wrong folder (just `src/cortex-m4` instead of the right `src/cortex-m4/fpv4-sp-d16-hard` in this case).