platformio/platformio-core #4455: Feature Request: option to stop LDF from running at every build

hamishcunningham, 2022-11-08:
Could an option be added to `pio run` to  turn off LDF for a build?

If I know that dependencies haven’t changed, I’d like not to take the time to run LDF by default, and so get a **faster edit/compile/flash** cycle.

(It doesn’t take too long, true, but when doing hundreds of builds during a coding session it adds up!)

Tnx!
H

@maxgerhardt confirmed that it [doesn't currently seem possible](https://community.platformio.org/t/prevent-ldf-from-running-at-every-build/30353) to do this.

ivankravets, 2023-11-28:
LDF typically needs <1sec to resolve project dependencies. If you have a lot of header files such as icons or fonts, it can slow down the resolving process. See example at https://community.platformio.org/t/scanning-dependencies-takes-more-than-90-seconds-every-build/36901/6?u=ivankravets

How many seconds does LDF take for your project?

sylque, 2024-03-19:
In my small project, pre-upload steps take **25 seconds**, although the project is already built.

Resolving dependencies takes 15 seconds (with 3 local libs and 5 github libs):
```
Processing m5StickCPlus2_debug (platform: espressif32; board: m5stick-c; framework: arduino)
-----------------------------------------------------------------------------------------------------------------
Library Manager: Installing symlink:../Fusion
Library Manager: Installing symlink:../Syl
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stick-c.html
PLATFORM: Espressif 32 (6.5.0) > M5Stick-C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoEigen @ 0.3.2+sha.8f073b5
|-- SparkFun BMI270 Arduino Library @ 1.0.2+sha.09d6aaf
|-- M5StickCPlus2 @ 1.0.2+sha.de5ea6c
|-- HardwareBLESerial-ESP32 @ 1.0.0+sha.99e55e0
|-- ESP32 BLE Arduino @ 2.0.0
|-- syl
|-- sylimu
|-- ArxTypeTraits @ 0.3.1
```
And then this guy alone takes 10 seconds, although the project is already compiled and there's no compiler output:
```
Building in debug mode
```