platformio/platformio-core #4456: Hook after Clean

Wiz-IO, 2022-11-11:
I use BUILD_DIR to compile boot2_stage boot loader of Raspberry Pi Pico
but Clean clear only board folder...

I didn't find a solution for hook Clean (after)
and I can offer you a small patch:

base.py
```py
    def on_clean(self, dir):            # <---
        pass
```
        
piotarget.py
```py
    print("Done cleaning")
    return build_dir                    # <---
```

main.py
```py
if env.GetOption("clean") or is_clean_all:
    dir = env.PioClean(is_clean_all)    # <---
    env.PioPlatform().on_clean(dir)     # <---
    env.Exit(0)
```
    
platform.py
```py
class MyPlatform(PlatformBase):
    def on_clean(self, dir):
        print('ON_CLEAN', dir) 
# RESULT: 
#   ON_CLEAN .....\.pio\build\test-board
```

ivankravets, 2022-11-18:
You can override the [run](https://github.com/platformio/platformio-core/blob/develop/platformio/platform/_run.py#L47) method and catch the "clean" target.

We also extended PlatformIO Build API and it is possible to catch "clean" target in the PRE script. See https://github.com/platformio/platformio-core/commit/7e6cb84c879a3d5080bd53183f80575d10a45cd5

ivankravets, 2022-11-18:
Yes, you can test it with `pio upgrade --dev`.

Wiz-IO, 2022-11-18:
nice !
Thanks

Wiz-IO, 2022-11-21:
Ivan, 
am i doing something wrong?
pre-script is executed with Build, but not with Clean

```py
Import("env")

def on_clean_pre(source, target, env):
    print('ON CLEAN PRE')

def on_clean_post(source, target, env):
    print('ON CLEAN POST')

env.AddPreAction("clean", on_clean_pre)
env.AddPostAction("clean", on_clean_post)

print('--------SCRIPT-----------')
```

sarthaknimbalkar, 2023-05-09:
Hey @ivankravets, I'm new to open source and would like to make a contribution on this issue. Could I possibly give it a try?

zeel01, 2023-11-02:
This seems to remain unavailable. There are generated files in my project that I would to delete when I run the `clean` or `fullclean` action, but as far as I can tell these actions still don't trigger pre/post hooks. This seems like it should be a pretty common need - cleaning up additional stuff that PIO doesn't normally know about.

dbarwacz, 2024-07-10:
It seems that adding 
```
env.AddPreAction("clean", on_clean_pre)
env.AddPostAction("clean", on_clean_post)
``` 
has no effect despite looking ok. 
The workaround I found is to create custom target that does the clean and remembering not to use the default one, although it's not ideal.