platformio/platformio-core #4459: Feature Request: pio remote support code coverage reporting

gittinhubbed, 2022-11-14:
### Feature Request
I am trying to add Sonar to my GitLab pipeline to get code coverage reporting and analysis. This works very well when running `pio test --junit-output-path .` however that option is not availalbe to `pio remote test`. I would like to be able to run `pio remote test -e <env_name> --junit-output-path <path_to_coverage_report>`. There is no coverage reporting option that I have found for this scenario but I am fairly new to embedded testing and development so I haven't been exposed to all the tools that are available so there may be one but don't know of it. 

### Description of problem
I am using an Arduino MKR WiFi 1010 and have not been able to find an emulator for the Cortex-M0 processor it uses. I need to run the unit tests through `pio remote [options]` and also want to generate the same code coverage reports that I am able to produce locally. 

### Expected Results
Executing `pio remote test --junit-output-path <report_path>` generates a coverage report and the report is stored at <report_path> where it can be used by the coverage analysis tool.

ivankravets, 2022-11-14:
Thanks for the feature request. As a temporary solution, you can run "local Gitlab runner" on your side and use `pio test` instead of `pio remote test`. See https://docs.gitlab.com/runner/

peterus, 2023-02-07:
@ivankravets suggesting a self hosted runner is a bad idea. if somebody is hijecking the runner with a simple push-request the runner and the internal network is open for for the hacker.