platformio/platformio-core #4462: extends doesn't override options as it is described in documentation

stas-sl, 2022-11-16:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Mac OS 11.7.1

**PlatformIO Version** (`platformio --version`): 6.1.5

### Description of problem

Documentations [says](https://docs.platformio.org/en/stable/projectconf/section_env_advanced.html#extends):

> If multiple items specified in the extends field then only values from the **latter** one will be used in the final configuration.

But in reality if multiple sections contain same options, then the value will be taken from the **first** one, not the **last**.

#### Steps to Reproduce

Basically, just copy the example from the documentation:

```ini
[base]
platform = teensy
framework = arduino
board = teensy31

[lib_deps1]
lib_deps =
    monstrenyatko/ArduinoMqtt @ ^1.5.1
    bblanchon/ArduinoJson @ ^6.18.3
    knolleary/PubSubClient @ ^2.8

[lib_deps2]
lib_deps =
    sstaub/Ticker @ ^4.3.0
    thijse/ArduinoLog @ ^1.1.1
    adafruit/RTClib @ ^1.14.0

[env:final]
extends = base, lib_deps1, lib_deps2
; The values from `lib_deps2` will override the values from lib_deps1, so
; the final `lib_deps` option will be equivalent to the following:
; lib_deps =
;    sstaub/Ticker @ ^4.3.0
;    thijse/ArduinoLog @ ^1.1.1
;    adafruit/RTClib @ ^1.14.0
```

and run `pio project config`:
```
> pio project config
Computed project configuration for .../test-extends
base
----
platform   =  teensy
framework  =  arduino
board      =  teensy31

lib_deps1
---------
lib_deps  =  monstrenyatko/ArduinoMqtt @ ^1.5.1
             bblanchon/ArduinoJson @ ^6.18.3
             knolleary/PubSubClient @ ^2.8

lib_deps2
---------
lib_deps  =  sstaub/Ticker @ ^4.3.0
             thijse/ArduinoLog @ ^1.1.1
             adafruit/RTClib @ ^1.14.0

env:final
---------
extends    =  base
              lib_deps1
              lib_deps2
platform   =  teensy
framework  =  arduino
board      =  teensy31
lib_deps   =  monstrenyatko/ArduinoMqtt @ ^1.5.1
              bblanchon/ArduinoJson @ ^6.18.3
              knolleary/PubSubClient @ ^2.8
```

### Actual Results

You can see, that lib_deps option value is taken from lib_deps1 section which comes first in the extends list.

### Expected Results

Options from latter sections overwrite options from previous sections as it is stated in the documentation.

stas-sl, 2022-11-17:
It was also mentioned before here: https://github.com/platformio/platformio-core/issues/3952.

ivankravets, 2023-01-14:
Thanks for the report. Please re-test with `pio upgrade --dev`.

closed 2023-01-14 by ivankravets

stas-sl, 2023-01-15:
Thanks for fixing it! I tried and it works now. 
💙💛

cyberman54, 2023-01-19:
i can confirm that issue #4448 is fixed on my windows machine with current --dev version, too.

commit fb5e9947 (2023-01-14): Fixed an issue when extends does not override options in the right order // Resolve #4462