platformio/platformio-core #4467: LDF can't install Python packages with spaces in the interpreter path

Saad-Mufti, 2022-11-21:
What kind of issue is this?

- [ X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**: Windows 11 Home 21H2

**PlatformIO Version** (`platformio --version`): 6.1.5

### Description of problem
The LDF (Library Dependency Finder) does not appear to be able to handle spaces in a user path when installing Python packages. I found this out when compiling for a project (platform: teensy) 

#### Steps to Reproduce

1. Create a PIO project (PIO must be installed on a path with a space in it)
2. Add a dependency that requires installation of some Python package (in my case, `Nanopb`)
3. Compile (`pio -v`)

### Actual Results
(Does not compile due to LDF splitting Python interpreter path because of a space character)

### Expected Results
Should compile

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:teensylc]
platform = teensy
board = teensylc
check_tool = cppcheck
framework = arduino
lib_deps =
  Nanopb
```
(The platform shouldn't matter, but I was using `teensy`.)

**Source file to reproduce issue:** **N/A**

### Additional info
Snippet of error trace:

```
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
C:\Users\Saad Mufti\.platformio\penv\Scripts\python.exe -m pip install "protobuf>=3.19.1"
'C:\Users\Saad' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1
```

Saad-Mufti, 2022-11-21:
I also came across [this](https://community.platformio.org/t/space-in-user-name-in-file-path-causes-error/28838/2) post on the platformio forums, which seems to be the exact same issue I am facing. There also appears to be a temporary fix, but it seems tedious to install the Python dependencies yourself especially in larger projects.

ivankravets, 2022-11-21:
Fixed in https://github.com/nanopb/nanopb/pull/834

closed 2022-11-21 by ivankravets