platformio/platformio-core #4468: Platformio build raising an unexpected error

PavelKosnar, 2022-11-21:
Hello, I get this error when trying to build my project in platformio:

Error: Traceback (most recent call last):
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\run\cli.py", line 144, in cli
    process_env(
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\run\cli.py", line 201, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\run\processor.py", line 83, in process
    install_project_env_dependencies(
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_libraries(project_env, options),
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 204, in _install_project_env_libraries
    _uninstall_project_unused_libdeps(project_env, options)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 280, in _uninstall_project_unused_libdeps
    lm.uninstall(spec)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\manager\_uninstall.py", line 29, in uninstall
    return self._uninstall(spec, skip_dependencies)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\manager\_uninstall.py", line 34, in _uninstall
    pkg = self.get_package(spec)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 263, in get_package
    spec = self.ensure_spec(spec)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 133, in ensure_spec
    return spec if isinstance(spec, PackageSpec) else PackageSpec(spec)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 184, in __init__
    self._parse(self.raw)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 291, in _parse
    raw = parser(raw)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 316, in _parse_requirements
    self.requirements = tokens[1].strip()
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 231, in requirements
    else semantic_version.SimpleSpec(str(value))
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\semantic_version\base.py", line 647, in __init__
    self.clause = self._parse_to_clause(expression)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1043, in _parse_to_clause
    return cls.Parser.parse(expression)
  File "C:\Users\kosna\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1063, in parse
    raise ValueError("Invalid simple block %r" % block)
ValueError: Invalid simple block ''

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

I've already tried to find a solution, but nothing seemed to work for me.

Tried:
     pip install -U platformio
     pio upgrade --dev
     reinstalling platformio
     reinstalling vs code
     reinstalling python
     removing "~/.platformio/penv" directory

maxgerhardt, 2022-11-22:
>  raise ValueError("Invalid simple block %r" % block)
> ValueError: Invalid simple block ''

What's the `platformio.ini`? Looks like an error in `lib_deps` to me, or in one of the included libraries.

PavelKosnar, 2022-11-23:
I just tried to delete the lib_deps and it's working. So the problem is with the lib_deps as you said. For some reason, no matter what I type in the lib_deps, it won't work.

maxgerhardt, 2022-11-23:
Delete the `.pio` folder of the project and set the `platformio.ini` as
```ini
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
monitor_speed = 115200
#platform_packages = platformio/framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
lib_deps =
  me-no-dev/ESPAsyncTCP@^1.2.2
  knolleary/PubSubClient@^2.8
  me-no-dev/ESP Async WebServer@^1.2.3
```
Does it work better?

PavelKosnar, 2022-11-24:
It's working now, thank you very much

closed 2022-11-25 by ivankravets