platformio/platformio-core #4469: Proposal to improve platformio.ini

softwaregravy, 2022-11-21:
- [X] **Feature Request**.

# Background

As my current project ramps up from tiny to small, I am looking for a more power config system. I see others with similar concerns. #3788 #4462 #3952 #4451 

As I see it, the current model is _override_ (or _replace_). Simply put, if a duplicate key is found the last one loaded wins and replaces all the options. The other issues and feature requests are looking for a _merge_ option which combines other entries found. 

To illustrate, consider the case of our code running in 2 modes: `staging` and `production`, and on 2 boards: `esp32_s2` and `esp32_s3`. These two each have configs. And these configs makes sense across all combinations. 

# Current

```ini
[staging]
build_flags = -DStaging
[production]
build_flags = -DProduction
[esp32_s2]
build_flags = -DS2
[esp32_s3]
build_flags = -DS3
```

Currently extending directly via `extends` or indirectly via `[env:` prefix _overrides_ values. So we have to currently include all previous flags by referencing them on a per-key basis

```ini
[s2_staging]
build_flags = ${staging.build_flags} ${esp32_s2.build_flags}
[s2_production]
build_flags = ${production.build_flags} ${esp32_s2.build_flags}
[s3_staging]
build_flags = ${staging.build_flags} ${esp32_s3.build_flags}
[s3_production]
build_flags = ${production.build_flags} ${esp32_s3.build_flags}
```

Hopefully my contrived example accurately illustrates the current status quo. 

# Proposal

I propose adding a an option that operates like `extends`, but it _merges_ instead of _overrides_. For now, lets call this `extend_merge` (but we can call it whatever). In this case, we'd still have the first part. 

```ini
[staging]
build_flags = -DStaging
[production]
build_flags = -DProduction
[esp32_s2]
build_flags = -DS2
[esp32_s3]
build_flags = -DS3
```

But then the 2nd part would reference this directly without knowing the keys:

```ini
[s2_staging]
extend_merge = staging, esp32_s2
[s2_production]
extend_merge = production, esp32_s2
[s3_staging]
extend_merge = staging, esp32_s3
[s3_production]
extend_merge = production, esp32_s3
```

In my little contrived case, the lines of code saved are minimal. But it is easy to imagine a case where the number of keys set in `staging` is 6-10, and then we'd save all those other lines.

Jason2866, 2022-11-24:
What should happen if there are duplicate but different entrys defined?

softwaregravy, 2022-11-24:
I don't follow what you mean by "duplicate but different"

ivankravets, 2022-11-24:
Duplicates of https://github.com/platformio/platformio-core/issues/3788

I'll provide more details in https://github.com/platformio/platformio-core/issues/3788

closed 2022-11-24 by ivankravets