platformio/platformio-core #4478: [feature request] cli: add the "build" target

MacDada, 2022-11-28:
### Configuration

**Operating system**: `MacOS 12.6.1 (21G217)`

**PlatformIO Version** (`platformio --version`): `6.1.5`

### Description of problem

I couldn't find a way to build the project from CLI [I was using IDE features before that]. [Turns out that simple `pio run` is enough](https://community.platformio.org/t/there-is-no-simple-build-target-to-run/30564).

### Actual Results

```
✗ pio
Usage: pio [OPTIONS] COMMAND [ARGS]...

Options:
  --version          Show the version and exit.
  -c, --caller TEXT  Caller ID (service)
  --no-ansi          Do not print ANSI control characters
  -h, --help         Show this message and exit.

Commands:
  access    Manage resource access
  account   Manage PlatformIO account
  boards    Board Explorer
  check     Static Code Analysis
  ci        Continuous Integration
  debug     Unified Debugger
  device    Device manager & Serial/Socket monitor
  home      GUI to manage PlatformIO
  org       Manage organizations
  pkg       Unified Package Manager
  project   Project Manager
  remote    Remote Development
  run       Run project targets (build, upload, clean, etc.)
  settings  Manage system settings
  system    Miscellaneous system commands
  team      Manage organization teams
  test      Unit Testing
  upgrade   Upgrade PlatformIO Core to the latest version
```

```
✗ pio run --list-targets
Environment    Group     Name         Title                        Description
-------------  --------  -----------  ---------------------------  ----------------------

nodemcuv2      Platform  buildfs      Build Filesystem Image
nodemcuv2      Platform  erase        Erase Flash
nodemcuv2      Platform  size         Program Size                 Calculate program size
nodemcuv2      Platform  upload       Upload
nodemcuv2      Platform  uploadfs     Upload Filesystem Image
nodemcuv2      Platform  uploadfsota  Upload Filesystem Image OTA
```

### Expected Results

A new target called `build`, so that we can run `pio run -t build`.

It would be an alias to `pio run` (without target specified) or the other way around, not specifying a target, would just call the `build` target.

### Additional info

Notice, that when I run `pio` it shows the line `  run       Run project targets (build, upload, clean, etc.)` – this is confusing, as I was expecting the `build` target to exist as listed.

hoeken, 2023-10-31:
Is there any progress on this?

Running pio run will compile the code, but it doesn't generate the firmware binary.  All I'm getting in my .pio/build folder is the firmware.elf.  If I run the "upload" target, then it will generate the firmware.bin file.  I'm trying to write a script to generate binaries for multiple environments to make firmware binaries for OTA so it would be great if I could automate this pls.

AgainPsychoX, 2024-05-24:
Bump; it can be a bit confusing indeed. 

Or at very least, add some note or something to the list displayed with `pio run --list-targets`:
```
PS D:\Projects\+embedded\picOOscilloscope> pio run -e pico --list-targets
Environment    Group     Name      Title                    Description
-------------  --------  --------  -----------------------  -------------
pico           Platform  buildfs   Build Filesystem Image
pico           Platform  uploadfs  Upload Filesystem Image
```

[Current docs](https://docs.platformio.org/en/latest/core/userguide/cmd_run.html#description) doesn't mention that `pio run` does the building. Also, why doesn't it list built-in targets?

anguis-datura, 2025-03-25:
So as far as I understand there is no way to just build filesystem and build firmware calling pio run once?