platformio/platformio-core #4480: Invalid "Can not remove temporary directory" error message

nomis, 2022-11-28:
### Configuration

**Operating system**: Ubuntu 18.04

**PlatformIO Version** (`platformio --version`): 6.1.4

### Description of problem

I get the following error message on every build:
```
Can not remove temporary directory `/.../aurora-coriolis/.pio/build`. Please remove it manually to avoid build issues
```

This code in `platformio/run/cli.py` completely ignores the exception that is generated, which has nothing to do with a build directory:
```
        # clean obsolete build dir
        if not disable_auto_clean:
            build_dir = config.get("platformio", "build_dir")
            try:
                clean_build_dir(build_dir, config)
            except:  # pylint: disable=bare-except
                click.secho(
                    "Can not remove temporary directory `%s`. Please remove "
                    "it manually to avoid build issues" % build_dir,
                    fg="yellow",
                )
```

The actual exception is:
```
  File "/home/simon/.local/lib/python3.7/site-packages/platformio/project/config.py", line 322, in _re_interpolation_handler
    assert parent_section.startswith("env:")
```

#### Steps to Reproduce

1. git clone https://github.com/nomis/aurora-coriolis/
2. cd aurora-coriolis
3. git checkout 2a1cc02d63c7bca1eb8329e1d6f3e1f1cdd69ef8
4. git submodule sync
5. git submodule update --init --depth 1
6. platformio run -e native
7. platformio run -e native

### Actual Results

```
Can not remove temporary directory `/.../aurora-coriolis/.pio/build`. Please remove it manually to avoid build issues
```

### Expected Results

No request to delete the `.pio/build` directory.

Error message about the "invalid" configuration?

### Additional info

The configuration in `/platformio.ini` and `/src/app/pio/config.ini` is complicated but it's not invalid.

The `test_testing_command` in the `app:native_test` section works correctly when used for testing environment `native_test`.

ivankravets, 2023-01-13:
Thanks for the report! Please re-test with `pio upgrade --dev`.

closed 2023-01-13 by ivankravets

nomis, 2023-01-25:
Yes that fixes the warning... but now I'll need to raise an issue for the invalid error that completely stops my build:
```
Error: `${this.__env__}` is called from the `app:native_test` section that is not valid PlatformIO environment, see for option `__env__` in section [app:native_test]
```

ivankravets, 2023-01-25:
> `app:native_test`

The `app:native_test` is not a PlatformIO env. It must start with `env:` prefix.

nomis, 2023-01-25:
No it doesn't. It must be evaluated in the context of an environment but it can be defined in a section that an environment later `extends`.

ivankravets, 2023-01-25:
`${this}` points to the current INI section where it was declared. Could you provide an example of using `this.__env__` and what do you expect?

nomis, 2023-01-25:
(The structure of the repo is a bit unusual because I'm using source from a submodule and only one `src_dir` is allowed. I need to put reusable environment config in the submodule `src/app/pio/config.ini` without those being defined as environments. The `platformio.ini` in the repo itself defines one or more environments.)

https://github.com/nomis/aurora-coriolis/blob/main/platformio.ini
```
[platformio]
extra_configs = src/app/pio/config.ini

[env:native_test]
extends = app:native_test
```

https://github.com/nomis/mcu-app/blob/63bff2e018c6f1c4433ba830693bde7b6105a839/pio/config.ini
```
[app:common]

[app:native]
extends = app:common

[app:native_test]
extends = app:native
test_testing_command =
	${platformio.src_dir}/app/pio/testing-command.sh
	${platformio.build_dir}/${this.__env__}/test
	${platformio.build_dir}/${this.__env__}/program
```

Running `platformio test -e native_test` executes `testing-command.sh` with the correct values (otherwise it wouldn't be able to execute the tests).

commit 17360b0e (2023-01-13): Show the real error message instead of "Can not remove temporary directory" when "platform.ini" is broken // Resolve #4480