platformio/platformio-core #4483: Only explicitly #include'd files should be built

MacDada, 2022-12-01:
Hi,

As far as I understand, if at least one file from "library" is `#include`d, all files from the library get built.

That is a problem for at least 2 reasons:

1. Slow builds – we build what we don't need, we waste time on that.
2. Compatibility issues
  * The file might not be "ready" or "complete" – it should not break the build, unless I'm actually using it.
  * The file might require some other file, that is not available [from external library, for example]. If I'm not using the file, why would I need to require that other lib it requires?
  * Makes testing [TDD workflow especially] harder

The real world example:

```
/lib/FooLib/FileA.cpp – includes `FileA.h` ofc
/lib/FooLib/FileA.h – includes `Servo.h`
/lib/FooLib/FileB.cpp – includes `FileB.h` ofc
/lib/FooLib/FileB.h – includes nothing
/src/main.cpp – includes `FileB.h`
```

From this structure, my build fails, because it does not find `Servo.h`. Even though I did not `#include` `Servo.h` anywhere. It is included in `FileA`, but I'm not using this file, so it should not break the build.

---

BTW, I did not check this: do all the files from `FooLib` get uploaded to my embedded device? Even those files I did _not_ `#include`?

---

The thread on the forum: https://community.platformio.org/t/problem-with-native-unit-testing-on-a-large-r-project/26683

MacDada, 2022-12-01:
If you're worried about backwards compatibility, a config option would be nice.

That's what I expected from `chain+` actually, that it would analyse `#include`s and load the files. It does. But it loads too many of them, even those not `#included` by my app.

ivankravets, 2022-12-07:
Duplicate of https://github.com/platformio/platformio-core/issues/3113

closed 2022-12-07 by ivankravets