platformio/platformio-core #4493: pio dependency installation on linux compiles all packages from source, ignoring wheels in pypi and piwheels

yschroeder, 2022-12-15:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**:
Debian GNU/Linux 11 (bullseye) on a Raspberry Pi 3B

**PlatformIO Version** (`platformio --version`):
6.1.5

### Description of problem
When running `pio remote agent start`, dependencies are installed via `pip`. Due to

https://github.com/platformio/platformio-core/blob/6395a032e5dfb7ad444718ec0da08e0f9370f3c0/platformio/package/manager/core.py#L150-L151

all packages are ALWAYS compiled from source which is extremely slow and annoying especially on slow hardware. It ignores all wheels in both, the pypi and piwheels index.

#### Steps to Reproduce

1. Run `pio remote agent start`
2. Wait an eternity for all packages to compile
3. Have everything recompiled when pio decided to update something

### Actual Results
`pip` compiles all packages from source.

### Expected Results
`pip` uses wheels when available.

### Workaround
Comment out both lines mentioned above. `pip` uses wheels for everything and installation is very fast.

yschroeder, 2022-12-15:
Digging deeper, this change was introduced to fix https://github.com/platformio/platformio-core/issues/3652.

However, I managed to use the remote agent on a Raspberry Pi with the wheel of `cryptography`, so maybe the issue is fixed?

yschroeder, 2023-02-16:
@ivankravets you marked this as known issue. Do you have a link to the other issue?

Is there anything I can do to help here? The issue still exists in version 6.1.6.

ivankravets, 2023-04-27:
Now dependencies will be built on the target side. Please re-test with `pio upgrade --dev`. Does it work now?

commit 2c94fb2a (2023-04-27): Resolved installation issues with PIO Remote on Raspberry Pi and other small form-factor PCs // Resolve #4425 , Resolve #4493 , Resolve  #4607
closed 2023-04-27 by ivankravets