platformio/platformio-core #4497: Projectpath broken cause if One Drive

Dork57, 2022-12-24:
hello,

I activeted OneDrive vor my projects path Now, I get a failure Message on compile. 
Knows anybody what to do is?

I removed \.pio\build manuelly but it stays the them issue.

`
io run -t menuconfig                                            
Can not remove temporary directory `C:\Users\Dirk\OneDrive\Dokumente\PlatformIO\Projects\221006-133048-espidf-arduino-blink\.pio\build`. Please remove it manually to avoid build issues
Processing esp32dev (board: esp32dev; platform: espressif32; framework: arduino, espidf)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden: 'C:\\Users\\Dirk\\OneDrive\\Dokumente\\PlatformIO\\Projects\\221006-133048-espidf-arduino-blink\\.pio':
  File "C:\Users\Dirk\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    os.makedirs(env.subst("$BUILD_DIR"))
  File "C:\Users\Dirk\.platformio\python3\lib\os.py", line 215:
    makedirs(head, exist_ok=exist_ok)
  File "C:\Users\Dirk\.platformio\python3\lib\os.py", line 215:
    makedirs(head, exist_ok=exist_ok)
  File "C:\Users\Dirk\.platformio\python3\lib\os.py", line 225:
    mkdir(name, mode)
================================================================= [FAILED] Took 1.03 seconds =================================================================
`

Dork57, 2022-12-24:
No new Testproject can be with create with the platformio guide. The message is: Could not initialize project and link to this forum.

Dork57, 2022-12-24:
If I want compile an other projects this is a part of the screenoutput

`
lease manually remove the file `C:\Users\Dirk\OneDrive\Dokumente\PlatformIO\Projects\BLE_Test\.pio\build\project.checksum`
[WinError 5] Zugriff verweigert: 'C:\\Users\\Dirk\\OneDrive\\Dokumente\\PlatformIO\\Projects\\BLE_Test\\.pio\\build'
Please manually remove the file `C:\Users\Dirk\OneDrive\Dokumente\PlatformIO\Projects\BLE_Test\.pio\build`
Can not remove temporary directory `C:\Users\Dirk\OneDrive\Dokumente\PlatformIO\Projects\BLE_Test\.pio\build`. Please remove it manually to avoid build issues
Processing esp32dev (board: esp32dev; framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "C:\Users\Dirk\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\Dirk\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__`

Dork57, 2022-12-25:
If I shift the project folder from OneDrive to a not OneDrive folder, all works like bevor. 
I think Microsoft still has one working.

dzid26, 2023-04-12:
I use it with OneDrive and get only occasional file conflicts when building.
But I actually store my projects in `C:\User\dzid26\Repositories` and then I just have a [symlink](https://schinagl.priv.at/nt/hardlinkshellext/linkshellextension.html) to it in My Documents which is synced to OneDrive. I don't know if that is what helps with this particular issue.

ivankravets, 2023-04-21:
Resolved in https://github.com/platformio/platformio-core/commit/743fc8e6365bb0b87163d012a7de397e1cec4495

Please re-test with `pio upgrade --dev`. Does it work now?

closed 2023-04-21 by ivankravets