platformio/platformio-core #4500: Support more Espressif boards

ChrisJones79, 2022-12-28:
**Feature Request for Custom Boards**

I've had issues trying to add a custom board. I followed the [instructions for adding a custom board](https://docs.platformio.org/en/stable/platforms/creating_board.html) as best I could but I have doubts about some of what I put in.

It would be really, really great if there were a tab or similar, accessible from PIO Home or the menu bar that brought up a Form that could populate a custom board json file. Additional helpful functionality would include:

- Ability to load values from existing custom board jsons
- Error checking in some form which could be extended based on usage patterns
- Creating a location, if needed, for the file to be saved to and saving it there
- Possibly connecting to repo of community made boards

###

ivykoko1, 2023-01-12:
I would also like a feature like this,a  community driven board manager, or support for installing arduino IDE type packages and convert them to platformio. I use some new and some less known boards that work 1 click in Arduino IDE but need 100 steps to correctly set up the board, platform and packages for it.

ChrisJones79, 2023-01-15:
A very good example would be the ESP32 devkits as it has several configurations/variants which all come more or less in the same package.

Series for the ESP32 -S2 -S3 -H -C2 -C3 -C6, etc. Which can vary in amount of flash and PSRAM. 

Currently I think only ~3 boards in a single variant are available?

tlangdon12, 2023-01-20:
I would also like more of the Espressif Devkit Boards to be supported, especially the ESP32-DevKitM-1 board, but anything that makes it easier to take an existing ESP32-Devkit board and convert it would be good.

PlastiBots, 2023-04-28:
I echo this request.  Namely ESP32S3 boards (dev flavour).  Specifically the LilyGo T-RGB board.   Note that the platform.ini file they provide on their github does not work for me.   Also, would like to see more clarity on use of PSRAM partitions.   My board has 16MB and the option board_build.partitions = default_16MB.csv does not work.  Only this works:  board_build.partitions = max_app_8MB.csv