platformio/platformio-core #4501: compile merlin with Visual Studio 6.1.5

williamlynn, 2022-12-29:
I am trying to compile a Merlin project using Visual Studio 6.1.5 on Windows 10.  This project compiled OK in previous versions of Visual Studio(8 months ago). I finally got the file to compile OK but when I copy it to another folder I get the following error code.

QUOTE
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/../feature/pause.h:33:10: fatal error: ../inc/MarlinConfigPre.h: No such file or directory
 #include "../inc/MarlinConfigPre.h"

          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\rambo\src\src\gcode\feature\trinamic\M906.cpp.o
*** [.pio\build\rambo\src\src\gcode\feature\runout\M412.cpp.o] Error 1
UNQUOTE

the following is the #include fro M906.cpp

QUOTE
#include "../../../inc/MarlinConfig.h"

#if HAS_FILAMENT_SENSOR

#include "../../gcode.h"
#include "../../../feature/runout.h"
UNQUOTE

The error message includes  MarlinConfigPre.h but the m906.h file has MarlinConfig.h. Hoe is the "Pre" getting added and how do I prevent it.

Bill Lynn

williamlynn, 2022-12-31:
It might be a Windows 10 issue or a bad VISUAL STUDIO install; I loaded Visual Studio onto a MAC(VENTURA) and the program compiled went without a problem

closed 2023-01-04 by ivankravets

AlexKBR2, 2023-05-06:
Hi all
 Have already 2 years i`m trying to make my btt skr mini e3 v2.0 and the tft35 v3.0.1 work together, but i do not suceed...
I always get a compilation error, this time, is this one:
 *  Executando tarefa: C:\Users\salex\.platformio\penv\Scripts\platformio.exe run 

Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Processing STM32F103RE_btt_USB (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 0.0.0
 - framework-cmsis @ 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|   |-- SoftwareSerial @ 1.0
|   |-- SPI @ 1.0
|-- U8glib-HAL @ 0.5.2
|   |-- Wire @ 1.0
|   |-- SPI @ 1.0
|-- IWatchdog @ 1.0.0
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- Servo @ 1.1.2
|-- Wire @ 1.0
Building in release mode
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\feature\runout\M412.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\feature\trinamic\M122.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\feature\trinamic\M569.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\feature\trinamic\M906.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\feature\trinamic\M911-M914.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\feature\trinamic\M919.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\gcode.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\gcode_d.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\geometry\G92.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\geometry\M206_M428.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\host\M110.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\host\M113.cpp.o
Compiling .pio\build\STM32F103RE_btt_USB\src\src\gcode\host\M114.cpp.o
In file included from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/marlinui.h:44,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/trinamic.h:35,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/indirection.h:44,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper.h:47,
                 from Marlin\src\gcode\feature\runout\../../../feature/runout.h:31,
                 from Marlin\src\gcode\feature\runout\M412.cpp:28:
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/../feature/pause.h:33:10: fatal error: ../inc/MarlinConfigPre.h: No such file or directory
   33 | #include "../inc/MarlinConfigPre.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RE_btt_USB\src\src\gcode\feature\runout\M412.cpp.o] Error 1
==================================================================================================================== [FAILED] Took 11.37 seconds ====================================================================================================================

Environment          Status    Duration
-------------------  --------  ------------
STM32F103RE_btt_USB  FAILED    00:00:11.373

Can some one help me with this, please?