platformio/platformio-core #4505: Feature request: provide some default flags if setting option for that in the platformio.ini file

jerabaul29, 2023-01-03:
I often end up having snippets like this in my platformio.ini files:

```
build_flags =
    !echo "-DREPO_GIT_BRANCH="$(git branch --show-current)
    !echo "-DREPO_COMMIT_ID="$(git log | head -1 | cut -c8-)  ; this is non portable, unix only; would need a python "pre platformio script" to get this to work portably...
    !echo "-DCOMPILING_HOST_NAME="$(hostname)  ; this is non portable, unix only; would need a python script else
    !echo "-DCOMPILING_USER_NAME="$(whoami)  ; this is non portable, unix only; would need a python script else
```

This works perfectly fine for me. The only "issue" is that, as you can see, I am on linux, and I have set this in a way that will not work in windows / possibly mac. I do not want to myself do some work to make things platform agnostic (I do not have access to a windows machine to test for example), but still it would be really convenient for my users if things were platform agnostic.

Hence my question: would it be possible to provide, from platformio-core "natively", some sets of "pre_defined_flags" that could be enabled to provide some typical, often needed, build flags?

For example something like:

```
pre_defined_flags =
    COMMIT_ID  ; provide as platform agnostic flag the git commit id from which compile if a git repo, "NONE" else
    GIT_BRANCH  ; provide as platform agnostic flag the git branch from which compile if a git repo, "NONE" else
    HOSTNAME  ; provide as platform agnostic flag the hostname on which compiling
    USERNAME  ; provide as platform agnostic flag the username who compiles
    ; COMPILE_UNIX_TIMESTAMP  ; provide as platform agnostic flag the unix timestamp of compilation time; now disabled
```

I guess this would mean:
- setting up a few python scripts that are platform agnostic / have branching for the different platforms available, to generate each of the corresponding pieces of information as strings at the start of the build process
- checking the content of the .ini file and the pre_defined_flag fields
- provide as compile flags the pre defined flags that are actually listed there

I.e. when building such a project, the flag ```COMMIT_ID``` containing the commit id would be visible, but the flag ```COMPILE_UNIX_TIMESTAMP``` would not.

As a small note / to help macro n00bs like me, it can be useful in documentation to remind that, to use all any of these flags, what is needed in the program is then:

```cpp

// the usual "macro stringification magics"
#define XSTR(a) #a
#define STRINGIFY_CONTENT(a) XSTR(a)

// getting the content to a variable and then printing the variable
static constexpr char commit_id[] {STRINGIFY_CONTENT(REPO_COMMIT_ID)};
Serial.println(F(commit_id));

// printing directly
Serial.println(STRINGIFY_CONTENT(COMPILING_USER_NAME))
```

ivankravets, 2023-01-03:
You can solve this issue with:

- https://docs.platformio.org/en/latest/scripting/index.html
- https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_flags.html#stringification

Also, [the list of built-in variables](https://github.com/platformio/platformio-core/blob/develop/platformio/builder/main.py), where you can find `$UNIX_TIME`. Please note that adding `$UNIX_TIME` to the build flags will trigger PlatformIO Build System to recompile the project from scratch. The value changes every second...

jerabaul29, 2023-01-03:
Many thanks for the pointers @ivankravets  :) . Yes, I remembered about these resources (or some older version of them), and I was aware that this can be used to solve the need :) .

I guess my point would be that it would be easier on the user to provide a list of default, readily implemented, turn key flags that are commonly used, so that the user does not need to set it up themselves but can just write a few words in the ini file. At least for some of the users I work with, who have no background in embedded from before / maybe know a tiny bit of C but no python / are not used to automation, it would be very helpful :) .

jerabaul29, 2023-01-03:
(and even for advanced users, it would be faster than reading the full doc, writing and testing the script, etc :) ).