platformio/platformio-core #4507: PlatformIO fails to run on M1 Apple MBP

jmgurney, 2023-01-06:
What kind of issue is this?

- [ X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macOS Ventura 13.0.1

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 6.1.5`

### Description of problem

I installed platformio, but when I went to try to process/run/build a project, I got the following error:
```
Compiling .pio/build/STM32F1/FrameworkArduinoVariant/board.cpp.o
sh: /Users/[xxx]/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-g++: Bad CPU type in executable
```

#### Steps to Reproduce

1. Run on apple M1 hardware
2. Prep virtual environment: `python3.10 -m venv p; . ./p/bin/activate`
3. Install PlatformIO: `pip install -U platformio`
4. Checkout blueSCSI: `git clone https://github.com/erichelgeson/BlueSCSI`
5. Build it: `cd BlueSCSI && platformio run`

### Actual Results

See above:
```
[...]
Dependency Graph
|-- SdFat @ 2.2.0
|   |-- SPI @ 1.0
Building in release mode
Compiling .pio/build/STM32F1/FrameworkArduinoVariant/board.cpp.o
sh: /Users/[xxx]/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-g++: Bad CPU type in executable
Compiling .pio/build/STM32F1/FrameworkArduinoVariant/wirish/boards.cpp.o
sh: /Users/[xxx]/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-g++: Bad CPU type in executable
Compiling .pio/build/STM32F1/FrameworkArduinoVariant/wirish/boards_setup.cpp.o
sh: /Users/[xxx]/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-g++: Bad CPU type in executable
Compiling .pio/build/STM32F1/FrameworkArduinoVariant/wirish/start.S.o
[...]
*** [.pio/build/STM32F1/FrameworkArduinoVariant/board.cpp.o] Error 126
sh: /Users/[xxx]/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-g++: Bad CPU type in executable
*** [.pio/build/STM32F1/FrameworkArduinoVariant/wirish/boards.cpp.o] Error 126
*** [.pio/build/STM32F1/FrameworkArduinoVariant/wirish/boards_setup.cpp.o] Error 126
[...]
=========================================================== [FAILED] Took 31.48 seconds ===========================================================

Environment    Status    Duration
-------------  --------  ------------
STM32F1        FAILED    00:00:31.482
====================================================== 1 failed, 0 succeeded in 00:00:31.482 ======================================================
```

### Expected Results

I expected that native arm binaries for the built tools are installed, and that the build would be successful.

### If problems with PlatformIO Build System:

Source and platformio.ini files are in the git repo cloned in the steps to reproduce.

### Additional info

ARM does make official Apple arm64 built tools available:
arm-gnu-toolchain-12.2.rel1-darwin-arm64-arm-none-eabi/bin/arm-none-eabi-gcc: Mach-O 64-bit executable arm64

from: https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

Yes, I could install rosetta 2 to run the x86_64 binaries, but platformIO should install native binaries for the hardware I'm running when available instead of non-native ones.  It is likely other people are not seeing this issue because they installed Rosetta 2 already.

jmgurney, 2023-01-06:
I just did some additional poking around, and it looks like your compatibility testing is incorrect:
https://registry.platformio.org/tools/platformio/toolchain-gccarmnoneeabi/compatibility?version=1.90301.200702

This lists that it should be compatible, but clearly installed the incorrect binaries, so likely the compatibility test system has rosetta 2 installed, meaning it cannot determine if packages are compatible with `darwin_arm64`.

I did just try the latest version, 1.100301.220327, and THAT does contain working `darmin_arm64` binaries.  It should refuse to install incompatible versions of the platform and issue a warning if they do not work.  Yes, it complicated by the fact that rosetta 2 may not be installed, but a simple test could be run, and an error issued about the incompatibility.

It looks like you only use the latest version to test compatibility, as https://registry.platformio.org/tools/platformio/toolchain-gccarmnoneeabi/compatibility?version=1.70201.0, which was released before the first M1 macs publicly available lists macos arm64 compatibility and we've already established that 1.90301.200702 is not compatible.

maxgerhardt, 2023-01-09:
There are tons of older binaries and toolchains which will likely not see recompilation for native Darwin ARM64 (think of: toolchain-atmelavr version 5.4.0), so I agree with the CEO there at [the community topic](https://community.platformio.org/t/cant-build-with-toolchain-atmelavr-on-darwin-arm64/26688/11?u=maxgerhardt) that Rosetta is needed.