platformio/platformio-core #4509: requirements for your system 'windows_amd64'

Fcscanf, 2023-01-07:
What kind of issue is this?

 **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:Windows 10

**PlatformIO Version** (`platformio --version`):6.1.5

### Description of problem
When installing a library, he told me that the system architecture is not supported

#### Steps to Reproduce

1.platformio.exe init --ide clion
2.
3.

### Actual Results
```
Resolving uno dependencies...
Library Manager: Installing mathworks/ThingSpeak @ ^2.0.0
Unpacking
Library Manager: ThingSpeak@2.0.0 has been installed!
Library Manager: Installing platformio/framework-arduinoespressif8266 @ ^3.30002.0
Error: Could not find the package with 'platformio/framework-arduinoespressif8266 @ ^3.30002.0' requirements for your system 'windows_amd64'

```
[image]

### Expected Results
The library was successfully installed

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:uno]
platform = atmelavr
board = uno
framework = arduino
upload_port = COM5
upload_speed = 115200
platform_packages =
    platformio/framework-arduino-avr@^5.1.0
lib_deps =
    mathworks/ThingSpeak@^2.0.0
    platformio/framework-arduinoespressif8266@^3.30002.0
```

### Additional info
```
>  pio system info                              

Python                      3.9.0-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp936
PlatformIO Core Directory   C:\Users\fcant\.platformio
PlatformIO Core Executable  D:\LinkSpace\Document\WorkSpace\Code\PyCharm\PythonProject\venv\Scripts\platformio.exe
Python Executable           D:\LinkSpace\Document\WorkSpace\Code\PyCharm\PythonProject\venv\Scripts\python.exe
Global Libraries            0
Development Platforms       2
Tools & Toolchains          10
--------------------------  --------------------------------------------------------------------------------------
```
The library related to ESP 8266 that is expected to be installed is officially available in Arduino, but there is no amd64 when viewing the architecture support in the platform warehouse
[image]
How can I solve this problem? Thank you for your review and help. Thank you

maxgerhardt, 2023-01-09:
The `platformio.ini` makes no sense, you are trying to include the entire Arduino ESP8266 **framework** as a library? What are you trying to achieve?

closed 2023-01-09 by ivankravets

rossog93, 2023-01-13:
I had the same issue when installing lennarthennigs/SimpleFSM @ ^1.2.0 library in `platfomio.ini`, with ESP32.

maxgerhardt, 2023-01-13:
The full `platformio.ini` being?

rossog93, 2023-01-13:
I followed the clues, the problem was python3 definition in my environment. Also, I noted 1.2.0 for SimpleFSM library is not available as default in Platformio, I don't understand why. Problems appears when I force it to the last version, from 1.0.0 (by default) to 1.0.2.
-Gabriel

Fcscanf, 2023-01-13:
@maxgerhardt 
I have an Arduino UNO development board and an ESP8266 Wifi module. At first, I referred to the Arduino hub project and used the AT command to achieve network connection and send data. But I also saw an example of using the ESP866Wifi library to operate the network connection. I wanted to use the library to operate the network connection, but I found that if the framework of using espressf8266 does not match the board of uno, Then the introduction of Framework-archinoespressf8266 as a library will not succeed.
The example I refer to is:[https://projecthub.arduino.cc/hbolanos2001/09032457-1c2d-4e75-9ab6-ee879275e8c6](https://projecthub.arduino.cc/hbolanos2001/09032457-1c2d-4e75-9ab6-ee879275e8c6)
Thank you for your help, thank you!