platformio/platformio-core #4510: Heltec ESP32 WiFi Kit V3 not supported

Eistee82, 2023-01-08:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Board Request
Hi I need pio support for Heltecs new ESP32 WiFi Kit V3.
It is a ESP32-S3 and it has difrent pins.
https://resource.heltec.cn/download/HTIT-WB32_V3(Rev1.1).pdf
https://resource.heltec.cn/download/WiFi_Kit_32_V3/HTIT-WB32_V3_Schematic_Diagram.pdf
https://resource.heltec.cn/download/WiFi_Kit_32_V3/HTIT-WB32_V3.png

Eistee82, 2023-01-08:
I think this is what I need: https://github.com/platformio/platform-espressif32/pull/954

ivankravets, 2023-01-09:
Duplicate of https://github.com/platformio/platform-espressif32/pull/954

closed 2023-01-09 by ivankravets