platformio/platformio-core #4514: Python errors?

shardin39, 2023-01-11:
What kind of issue is this?
While working on a program that was compiling correctly the following issues began.
Reinstalled current version PIO;
Installed pre-release version 3+ PIO;
Reinstalled/updated Python;

Same errors

Processing d1_mini_pro (platform: espressif8266; board: d1_mini_pro; framework: arduino)
---------------------------------------------------------------------------------------------------------------Removing unused dependencies...
Error: Traceback (most recent call last):
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\steve\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__    
    return self.main(*args, **kwargs)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main        
    rv = self.invoke(ctx)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\cli.py", line 71, in invoke    
    return super().invoke(ctx)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke      
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\steve\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke      
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\run\cli.py", line 144, in cli
    process_env(
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\run\cli.py", line 201, in process_env     
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\run\processor.py", line 83, in process    
    install_project_env_dependencies(
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_libraries(project_env, options),
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 204, in _install_project_env_libraries
    _uninstall_project_unused_libdeps(project_env, options)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 280, in _uninstall_project_unused_libdeps
    lm.uninstall(spec)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\manager\_uninstall.py", line 29, in uninstall
    return self._uninstall(spec, skip_dependencies)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\manager\_uninstall.py", line 34, in _uninstall
    pkg = self.get_package(spec)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 263, in get_package
    spec = self.ensure_spec(spec)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 133, in ensure_spec
    return spec if isinstance(spec, PackageSpec) else PackageSpec(spec)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 184, in __init__   
    self._parse(self.raw)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 291, in _parse     
    raw = parser(raw)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 316, in _parse_requirements
    self.requirements = tokens[1].strip()
  File "C:\Users\steve\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 231, in requirements
    else semantic_version.SimpleSpec(str(value))
  File "C:\Users\steve\.platformio\penv\lib\site-packages\semantic_version\base.py", line 647, in __init__     
    self.clause = self._parse_to_clause(expression)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1043, in _parse_to_clause
    return cls.Parser.parse(expression)
  File "C:\Users\steve\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1063, in parse       
    raise ValueError("Invalid simple block %r" % block)
ValueError: Invalid simple block ''

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

[... 168 lines cut ...]
  //Send raw text message
  message.text.content = Emsg.c_str();
  message.text.charSet = "us-ascii";
  message.text.transfer_encoding = Content_Transfer_Encoding::enc_7bit;
  
  message.priority = esp_mail_smtp_priority::esp_mail_smtp_priority_low;
  message.response.notify = (esp_mail_smtp_notify_success) | (esp_mail_smtp_notify_failure) | (esp_mail_smtp_notify_delay);

  
  /* Set the custom message header */
  message.addHeader("Message-ID: <abcde.fghij@gmail.com>");

 // Connect to server with the session config 
  if (!smtp.connect(&session)){return 9;}

  // Start sending Email and close the session 
  if (!MailClient.sendMail(&smtp, &message))
  {  
    Serial.println("Error sending Email, " + smtp.errorReason());
    return 99;
  }
  return 1;
}

### Additional info

ivankravets, 2023-01-11:
Do you have a broken dependency declared in platformio.ini?

shardin39, 2023-01-12:
Hi Ivan; 

Worked on this most of the day without results. Then, it started working. No idea why... 

Thanks for the reply. 

Steve 

Ϻολών λαβέ 

ab abusu ad usum non valet consequentia 

The only thing necessary for the triumph of evil is for good men to do nothing.” 

"Malo periculosam, libertatem quam quietam servitutem." T. Jefferson 

Ave atque vale!

ivankravets, 2023-01-12:
Please remove `.pio` folder in the project root. The problem should be fixed.

closed 2023-01-12 by ivankravets

shardin39, 2023-01-12:
Hi Ivan; 

I will keep that in mind should things go south again. 

Thanks! 

Steve