platformio/platformio-core #4520: my original error in HAL

jayhawk303, 2023-01-15:
What kind of issue is this?
S C:\Users\jayha\Desktop\3d print fw\Marlin-bugfix-2.1.x> pip install -U hal
Collecting hal
  Using cached HAL-0.0.3.zip (27 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [9 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\jayha\AppData\Local\Temp\pip-install-qnf3qiqq\hal_a60be58102154319b7cbf1e4320bf8b7\setup.py", line 2, in <module>
          import ez_setup
        File "C:\Users\jayha\AppData\Local\Temp\pip-install-qnf3qiqq\hal_a60be58102154319b7cbf1e4320bf8b7\ez_setup.py", line 106
          except pkg_resources.VersionConflict, e:
                                              ^
      SyntaxError: invalid syntax
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip
- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

closed 2023-01-23 by ivankravets

jayhawk303, 2023-01-30:
Error pops up asking me to report a bug, that's all I was trying to do

ivankravets, 2023-01-30:
Do you mean Marlin asks you? Please forward issue to https://github.com/MarlinFirmware/Marlin/issues