platformio/platformio-core #4523: Platform IO is garbage. Cant even use a Heltec_Wifi_32_v3

madscientest, 2023-01-16:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

peterhoeg, 2023-01-17:
If you want help, I suggest providing some additional information so that people can actually help you.

madscientest, 2023-01-17:
I didnt ask for help sir. I just want to bitch and complain about the shitty software.

peterhoeg, 2023-01-17:
In that case, please do that elsewhere. This place is for people who are interested in making pio better.

madscientest, 2023-01-17:
No, The only thing that makes PIO better is gas and a match.

peterhoeg, 2023-01-17:
@valeros , can we plesae just block this person?

madscientest, 2023-01-17:
LOL. Thanks for the laugh.

closed 2023-01-17 by ivankravets