platformio/platformio-core #4526: PlatformIO thinks there is no internet connection

niklasarnitz, 2023-01-19:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
- macOS Big Sur
- Arch Linux with the latest updates
**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.4
### Description of problem
PlatformIO thinks I don't have internet.

#### Steps to Reproduce

1. Create a NodeMCU ESP8266 Arduino Project
2. Upload the empty file

### Actual Results
```
 *  Executing task: platformio run --target upload --environment nodemcuv2_ota 

Processing nodemcuv2_ota (platform: espressif8266 @^2.6.3; board: nodemcuv2; framework: arduino)
---------------------------------------------------------------------------------------------------
Library Manager: Installing lebuni/ZACwire for TSic @ 2.0.0
Error: You are not connected to the Internet.
PlatformIO needs the Internet connection to download dependent packages or to work with PlatformIO Account.

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 'nodemcuv2_ota'" terminated with exit code: 1.
 ```

### Expected Results
It gets the toolchain and successfully builds and uploads it.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
lib_dir = lib
src_dir = src
extra_configs = platformio_extra.ini

[env]
lib_deps =
    lebuni/ZACwire for TSic @ 2.0.0
    schm1tz1/TSIC @ 1.1.2
    milesburton/DallasTemperature @ 3.11.0
    paulstoffregen/OneWire @ 2.3.7
    adafruit/Adafruit_VL53L0X @ 1.2.0
    olkal/HX711_ADC @ 1.2.12
    olikraus/U8g2 @ 2.34.5
    git+https://github.com/rancilio-pid/Arduino-PID-Library
    knolleary/PubSubClient @ 2.8.0
    me-no-dev/AsyncTCP @ 1.1.1
    bblanchon/ArduinoJson @ 6.19.4
    tobiasschuerg/ESP8266 Influxdb @ 3.9.0
    git+https://github.com/me-no-dev/ESPAsyncWebServer#f71e3d4
    git+https://github.com/tzapu/WiFiManager#71937d1

[env:nodemcuv2_usb]
platform = espressif8266 @^2.6.3
board = nodemcuv2
board_build.filesystem = littlefs
framework = arduino
monitor_speed = 115200
monitor_filters = esp8266_exception_decoder
debug_tool = esp-prog
debug_init_break = tbreak setup

[env:nodemcuv2_ota]
platform = espressif8266 @^2.6.3
board = nodemcuv2
board_build.filesystem = littlefs
framework = arduino
monitor_speed = 115200
monitor_port = socket://rancilio-silvia:23
upload_protocol = espota
upload_port = rancilio-silvia
upload_flags = --auth=otapass

[env:esp32_usb]
platform = espressif32 @^5
board = esp32dev
board_build.filesystem = littlefs
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
debug_tool = esp-prog
debug_init_break = tbreak setup

[env:esp32_ota]
platform = espressif32 @^5
board = esp32dev
board_build.filesystem = littlefs
framework = arduino
monitor_speed = 115200
upload_protocol = espota
upload_port = silvia
upload_flags = --auth=otapass
```

**Source file to reproduce issue:**
See https://github.com/rancilio-pid/clevercoffee

### Additional info
- I'm using Google DNS Servers
- I'm not using any Proxy
- I'm not using any network level AdBlocker

ivankravets, 2023-01-19:
Could you debug https://github.com/platformio/platformio-core/blob/develop/platformio/http.py#L183 ?

https://github.com/platformio/platformio-core/blob/6db3eb8e336b7530eff0886dec861f6015aced8f/platformio/__init__.py#L56

closed 2023-01-19 by ivankravets

niklasarnitz, 2023-01-19:
`88.198.170.159` is not reachable.
But the GitHub IP and Domain are.

niklasarnitz, 2023-01-19:
Using `socket.create_connection` never yields a positive result for me.

niklasarnitz, 2023-01-19:
```python
import socket

socket.setdefaulttimeout(2)

hosts = [
	"github.com",
	"185.199.110.153",  # Github.com
	"88.198.170.159",  # platformio.org
	"registry.platformio.org",
	"registry.nm1.platformio.org",
]

for host in hosts:
	print(host)
	s = socket.create_connection((host, 80))
	s.close()
```

Output:
```
github.com
Traceback (most recent call last):
  File "/Users/niklas/test.py", line 15, in <module>
    s = socket.create_connection((host, 80))
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
TimeoutError: timed out
```

niklasarnitz, 2023-01-19:
Okay, via celullar it works.

It seems that my ISP is blocking the platformio.org IP...

ivankravets, 2023-01-19:
Could you contact your ISP? What is the reason that they blocked PlatformIO?

niklasarnitz, 2023-01-19:
As I said, It seems they blocked it.
Bc on my end there is nothing blocked.
I’ll contact them.