platformio/platformio-core #4532: Vulnerability report

luigigubello, 2023-01-27:
Hi 👋 I need to report a potential vulnerability in PlatformIO Core, but you don't have a security policy. How should I proceed?

ivankravets, 2023-01-27:
Hi Luigi, please email us directly via contact@piolab.com or provide a PR. We will release the new version ASAP if the issue is critical. Thanks!

luigigubello, 2023-01-27:
Hi @ivankravets 👋 Unfortunately, I don't have enough knowledge on this project and the codebase seems too big to propose a good PR, but I am writing to contact@piolab.com to provide all info to reproduce the bug. Thank you!

luigigubello, 2023-01-27:
Hi @ivankravets
I have tried to send the e-mail, but I have received this message: 
```
554 5.7.1 <[contact@piolab.com](mailto:contact@piolab.com)>: Relay access denied
```
May you share another e-mail address or fix the server?

Thank you

ivankravets, 2023-01-27:
Thanks for the report. Please re-test with `pio upgrade --dev`.

P.S.: Yes, it should piolabs.com. Sorry for the typo :(

closed 2023-01-27 by ivankravets

luigigubello, 2023-01-27:
I can confirm that I cannot reproduce the issue using the same payload and proof-of-concept 👌 I have not tried to find regex bypass and I am still investigating PlatformIO 6.1.6 to check if there are other entry-points (or just other vulnerabilities :)). Will Platformio report the vulnerability in the [GitHub Security Advisories](https://github.com/platformio/platformio-core/security/advisories)?

commit 15d53c95 (2023-01-27): Prevent shell injection when converting INO file to CPP // Resolve #4532