platformio/platformio-core #4536: Failed to find registry package after installing it with pio lib install

cguimaraes, 2023-01-31:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.04 and MacOS 13.0.1 (Ventura)

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.6a4

### Description of problem
When installing Zenoh-Pico from PlatformIO registry (https://registry.platformio.org/libraries/cguimaraes/zenoh-pico), it fails to build because it cannot find Zenoh-Pico package.

Upon installing the library using `pio lib install cguimaraes/zenoh-pico`, the following entry in `platformio.ini` is added:

```ini
lib_deps = cguimaraes/zenoh-pico@^0.7.0-rc
```

While doing some investigation, it seems that the issue lies on having the `^` in `@^0.7.0-rc`.
Meaning, it works as expected if:
```ini
lib_deps = cguimaraes/zenoh-pico
```
or
```ini
lib_deps = cguimaraes/zenoh-pico@0.7.0-rc
```

Also, setting the GitHub works as expected:
```ini
lib_deps = https://github.com/eclipse-zenoh/zenoh-pico
```

#### Steps to Reproduce

1. pio init -b esp32-c3-devkitm-1
2. pio lib install cguimaraes/zenoh-pico
3. pio run 

### Actual Results

```bash
Library Manager: Installing cguimaraes/zenoh-pico @ ^0.7.0-rc
Error: Could not find the package with 'cguimaraes/zenoh-pico @ ^0.7.0-rc' requirements for your system 'linux_x86_64'
```

```bash
Library Manager: Installing cguimaraes/zenoh-pico @ ^0.7.0-rc
Error: Could not find the package with 'cguimaraes/zenoh-pico @ ^0.7.0-rc' requirements for your system 'darwin_arm64'
```

### Expected Results
```bash
(...)
Scanning dependencies...
Dependency Graph
|-- zenoh-pico @ 0.7.0-rc
|   |-- BluetoothSerial @ 2.0.0
(...)
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32-c3-devkitm-1]
platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino
lib_deps = cguimaraes/zenoh-pico@0.7.0-rc
```

**Source file to reproduce issue:**
An empty project is sufficient

### Additional info
The issue happens for any of the supported frameworks by Zenoh-Pico library: Arduino, ESPIDF, Zephyr and Mbed.

ivankravets, 2023-01-31:
We use [semantic-version](https://pypi.org/project/semantic-version/) Python package to operate with versions. Could you try to debug that module? Nevertheless, we don't recommend publishing beta software including release candidates to the registry. Developers can easily use these development branches with Github URLs. See https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#repository-git-hg-svn

Would you like us to delete all RC/Beta versions from the registry? Please publish the stable version and we will provide cleanup.

cguimaraes, 2023-02-01:
The versioning semantic seems to be compatible with [semantic-version](https://pypi.org/project/semantic-version/).
```
>>> v1 = semantic_version.Version('0.6.0-beta.1')
>>> v2 = semantic_version.Version('0.7.0-rc')
>>> list(v1)
[0, 6, 0, ('beta', '1'), ()]
>>> list(v2)
[0, 7, 0, ('rc',), ()]
>>> print(v1 < v2)
True
>>> print(v2 > v1)
True
```

The two versions published in the registry are tagged as `beta` and `rc`, simply due to Eclipse Foundation guidelines as the project is still in a rapid application development stage.

ivankravets, 2023-02-01:
```python
>>> s = semantic_version.SimpleSpec('^0.7.0-rc')
>>> v1 in s
False
>>> v2 in s
False
>>> v2 in s
```

What is interesting is that `semantic_version.NpmSpec` produces absolutely different results:

```python
>>> s2 = semantic_version.NpmSpec('^0.7.0-rc')
>>> v1 in s2
False
>>> v2 in s2
True
```

In PlatformIO Core, we use `semantic_version.SimpleSpec` to keep the declaration simple and understandable.

Could you report this issue to https://github.com/rbarrois/python-semanticversion/issues ?

cguimaraes, 2023-02-01:
Thanks @ivankravets .
I will report it there and tag this issue.

ivankravets, 2023-03-28:
Would you like us to delete all RC/Beta versions from the registry? Please publish the stable version and we will provide a cleanup.

cguimaraes, 2023-03-28:
The two versions published in the registry are tagged as beta and rc, simply due to Eclipse Foundation guidelines as the project is still in a rapid application development stage.

Still, they are stable releases.

ypearson-bdai, 2024-08-01:
Is this fixed yet?
`lib_deps = https://github.com/eclipse-zenoh/zenoh-pico#1.0.0.5`
doesn't install the right version

ivankravets, 2024-08-02:
Could you follow the specification documented in https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#repository-git-hg-svn ?