platformio/platformio-core #4537: Library Search Order

thjean, 2023-02-01:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 6.1.6

### Description of problem
The search order of the different storage folders where LDF searches for libraries does not seem to match the order described in https://docs.platformio.org/en/latest/librarymanager/ldf.html, section "Storage".

My observation is that libraries located in libdeps_dir have higher priority than libraries stored in lib_dir. 

#### Steps to Reproduce

1. create new project
2. add a library using the library manager in PIO Home. The lib is stored in folder libdeps as specified.
3. place a copy of the library in folder lib and do some changes in the copy. 
4. compile and run the project, watch which library has been used.   

### Actual Results
The library files as stored in libdeps folder are used.

### Expected Results
The library files as stored in lib folder are used. 

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**

[env:bluepill_f103c8_128k]
platform = ststm32
board = bluepill_f103c8_128k
framework = arduino
build_flags = -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
lib_deps = robtillaart/SHT31@^0.3.7

**Source file to reproduce issue:**

#include <Arduino.h>
#include <SHT31.h>

void setup()
{
  Serial.begin(115200);
}

void loop()
{
  Serial.print("SHT31_LIB_VERSION: \t");
  Serial.println(SHT31_LIB_VERSION);
  delay(100);
}

### Additional info
In my example above, I used the SHT31 Lib, placed a copy of SHT31.cpp and SHT31.h to the lib folder and changed the version string in the file lib\SHT31\SHT31.h. Running the example outputs the version string as defined in the libdefs file instead of the expected string as defined in the file stored in the lib folder.
Maybe I misunderstand the LDF search. I assume the libraries are searched and used according to the folder order as listed in https://docs.platformio.org/en/latest/librarymanager/ldf.html, even if there are copies located in the different library storage folders.

ivankravets, 2023-04-12:
The `lib_deps` option forces PlatformIO to ALWAYS include specified libraries in the build process. You can try to comment on that line and library from `lib` folder should be used according to https://github.com/platformio/platformio-core/blob/58a1d5d96e6a1259dcb5d12b7b103eff4bbdd1fd/platformio/builder/main.py#L131

thjean, 2023-04-12:
Thank you for the details on the lib_deps priority. Indeed, removing the lib_deps line from the .ini file leads to the lib_dir folder being used. 

Seems like I misunderstand the priority description in https://docs.platformio.org/en/latest/librarymanager/ldf.html#storage

ivankravets, 2023-04-12:
I reopen this issue and will update docs. Thanks

ivankravets, 2023-04-13:
I've just tried to dig into the code base. See what I found https://github.com/platformio/platformio-core/blob/66fe55668e8bab03e12377dd55613f6f8b11b9e5/platformio/builder/tools/piolib.py#L962:L974

So, it has correct behavior according to the docs. I also tried to reproduce your issue above and it works well for me. 

Are you sure that you copied WHOLE library folder from `libdeps` to the `lib` folder? Please note that you need to copy ALL files from the lib including `lib/SHT31/.piopm` file.

closed 2023-04-13 by ivankravets

thjean, 2023-04-13:
You've got me there :)

When doing the test, I just copied the `.cpp` and the `.h` files, I did not copy the `.piopm` file. Copying the `.piopm` file does the trick and the `lib` dir gets priority. 

Thank you for this valuable information and sorry for rising an unnecessary issue. 

Regards
Jean