platformio/platformio-core #4539: This IS NOT fixed

LeonaK, 2023-02-02:
This is NOT repaired. I have been trying to compile files that were compiling perfectly well for months. Now, NOthing compiles. I have reloaded 5.2.5,, no help. I updated to the newest version, no help. This is just awful. I imagine 3d printers all over the country stalled because of YOUR problem.
I understand that mistakes can be made, but to just go on for months, pretending that the problem is solved is shameful.

    Executing task in folder G:: C:\Users\klajd.platformio\penv\Scripts\platformio.exe run --target clean

*******************************************************************************************************************************************************Obsolete PIO Core v6.1.6 is used (previous was 6.1.7a1)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
*******************************************************************************************************************************************************Warning! Ignore unknown configuration option monitor_flags in section [env]
Warning! src_filter configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use build_src_filter instead
Warning! src_filter configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use build_src_filter instead
Warning! src_filter configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use build_src_filter
instead
Warning! src_build_flags configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use build_src_flags instead
Warning! src_filter configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use build_src_filter instead
Warning! src_build_flags configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use build_src_flags instead
Warning! src_filter configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use build_src_filter instead
Warning! src_filter configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use build_src_filter instead
Warning! src_filter configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use build_src_filter instead
Warning! src_filter configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use build_src_filter instead
Warning! src_filter configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use build_src_filter instead
Warning! src_filter configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use build_src_filter instead
Warning! src_filter configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use build_src_filter instead
Processing BIGTREE_SKR_2_F429 (board: marlin_STM32F429VGT6; platform: ststm32@~12.1; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via -v, --verbose option
PermissionError: [WinError 5] Access is denied: 'C:\Users\klajd\.platformio\packages\framework-arduinoststm32\variants\MARLIN_F4x7Vx\hal_conf_extra.h':
File "C:\Users\klajd.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
env.SConscript(item, exports="env")
File "C:\Users\klajd.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\klajd.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
File "G:\buildroot\share\PlatformIO\scripts\generic_create_variant.py", line 48:
shutil.rmtree(variant_dir)
File "C:\Users\klajd.platformio\python3\lib\shutil.py", line 749:
return _rmtree_unsafe(path, onerror)
File "C:\Users\klajd.platformio\python3\lib\shutil.py", line 627:
onerror(os.unlink, fullname, sys.exc_info())
File "C:\Users\klajd.platformio\python3\lib\shutil.py", line 625:
os.unlink(fullname)
============================================================= [FAILED] Took 3.62 seconds =============================================================

Environment Status Duration

BIGTREE_SKR_2_F429 FAILED 00:00:03.618
======================================================== 1 failed, 0 succeeded in 00:00:03.618 ========================================================

    The terminal process "C:\Users\klajd.platformio\penv\Scripts\platformio.exe 'run', '--target', 'clean'" terminated with exit code: 1.
    Terminal will be reused by tasks, press any key to close it.
    it.

maxgerhardt, 2023-02-02:
> PermissionError: [WinError 5] Access is denied: 'C:\Users\klajd.platformio\packages\framework-arduinoststm32\variants\MARLIN_F4x7Vx\hal_conf_extra.h':

How is this PlatformIO's fault when a script written by Marlin developers (`G:\buildroot\share\PlatformIO\scripts\generic_create_variant.py`) tells Windows to remove this directory and Windows returns an error?  Have you tried removing the directory manually?

> Warning! src_filter configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use build_src_filter instead

This indicates you are using an outdated version of Marlin, because it [has been changed](https://github.com/MarlinFirmware/Marlin/blame/32e8627510498da3a88d9e198f7c17a6a2ad8a66/platformio.ini#L285-L288) to `build_src_filter` 8 months ago.

>Obsolete PIO Core v6.1.6 is used (previous was 6.1.7a1)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html

You should also follow these steps to cleanup your system.

ivankravets, 2023-02-02:
Thanks, @maxgerhardt, for the correct answer.

closed 2023-02-02 by ivankravets