platformio/platformio-core #4542: pio debug does not honor default_envs

jmgurney, 2023-02-04:
What kind of issue is this?

- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: 
Ventura 13.2
```
Darwin xxx 22.3.0 Darwin Kernel Version 22.3.0: Thu Jan  5 20:48:54 PST 2023; root:xnu-8792.81.2~2/RELEASE_ARM64_T6000 arm64
```

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 6.1.5`

### Description of problem

When building debug, only the first environment listed in default_envs is built.  The documentation claims for `default_envs` (https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/generic/default_envs.html#projectconf-pio-default-envs) claims that debug uses it as well:
> Also, [Debugging](https://docs.platformio.org/en/latest/plus/debugging.html#piodebug) checks this option when looking for debug environment.

I expect that all the environments to be built as this helps ensure that there are not any build failures.  Having to build each debug build separately makes `default_envs` less useful, as now you might as well use the same way to build non-debug builds.

#### Steps to Reproduce

1.  Follow instructions at https://docs.platformio.org/en/stable/core/quickstart.html#core-quickstart to start a project.
2.  Run `pio run` to see that all three envs are built.
3.  Run `pio debug` and see that only the `uno` env is built

### Actual Results

```
Environment    Status    Duration
-------------  --------  ------------
uno            FAILED    00:00:00.265
========== 1 failed, 0 succeeded in 00:00:00.265 ==========

```

(not failure is due to not having Rosetta 2 installed, and does not matter, as it should list all three as failed.

### Expected Results

I would expect the same results as I get from `pio run`:
```
Environment    Status    Duration
-------------  --------  ------------
uno            FAILED    00:00:00.256
nodemcuv2      FAILED    00:00:00.350
teensy31       FAILED    00:00:00.336
========== 3 failed, 0 succeeded in 00:00:00.942 ==========
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:uno]
platform = atmelavr
board = uno
framework = arduino

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

[env:teensy31]
platform = teensy
board = teensy31
framework = arduino

```

**Source file to reproduce issue:**
```cpp
/**
 * Blink
 *
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */
#include "Arduino.h"

#ifndef LED_BUILTIN
#define LED_BUILTIN 13
#endif

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);

  // wait for a second
  delay(1000);

  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);

   // wait for a second
  delay(1000);
}

```

### Additional info

ivankravets, 2023-02-04:
You can do step-by-step debugging only with one environment. If you mean building (not debugging) ALL envs in [debug mode](https://docs.platformio.org/en/latest/projectconf/build_configurations.html), set `build_type = debug` and then `pio run`.

closed 2023-02-04 by ivankravets

jmgurney, 2023-02-05:
I'm confused, when I run pio debug, it only builds the firmware for debugging, it doesn't actually start debugging unless I specify an interface.

Is there a way to build all [default] environments for both debug and non-debug w/o having to modify `platformio.ini`?  I'm trying to make sure that changes don't break the different build environments w/o building out extra/unneeded scripts.

ivankravets, 2023-02-05:
If you use CLI, you can pass the custom [project-conf](https://docs.platformio.org/en/latest/core/userguide/cmd_run.html#cmdoption-pio-run-c). Example:

- platformio.ini
- platformio-debug.ini

**platformio.ini**

```ini
[env:uno]
platform = atmelavr
board = uno
framework = arduino

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
```

**platformio-debug.ini**

```ini
[platformio]
extra_configs = platformio.ini

[env]
build_type = debug
```

**Build in debug mode**

```
pio run --project-conf platformio-debug.ini
```