platformio/platformio-core #4543: Disable automatic dependency download!

zekageri, 2023-02-04:
I just realized that my project does not use the libraries i wanted.
I'm using a forked version of `AsyncTCP` and `ESP_AsyncWebServer`.
I use it as i always used, download from github, put it into lib folder.

I see that PIO just automatically downloaded the original libraries from `Me_no_dev` and a bunch of bugs and errors came back.

If i delete lib_deps folder, PIO will download them again even if i did not ask it to.

??

ivankravets, 2023-02-04:
What is your `platformio.ini` file?

zekageri, 2023-02-04:
```
; PlatformIO Project Configuration File 
 ; 
 ;   Build options: build flags, source filter 
 ;   Upload options: custom upload port, speed and extra flags 
 ;   Library options: dependencies, extra library storages 
 ;   Advanced options: extra scripting 
 ; 
 ; Please visit documentation for the other options and examples 
 ; https://docs.platformio.org/page/projectconf.html 
  
 [env:esp-wrover-kit] 
 platform = espressif32 
  
 board       = esp-wrover-kit 
 framework   = arduino 
 board_build.filesystem = littlefs 
  
 board_build.f_cpu   = 240000000L 
 upload_port     = COM3 
 upload_speed    = 921600 
 monitor_speed   = 115200 
 monitor_filters = esp32_exception_decoder 
  
 board_upload.flash_size = 16MB 
 board_build.flash_mode = qio 
 board_build.partitions = large_spiffs_16MB.csv 
 board_build.f_flash    = 80000000L 
  
 build_flags =   -DBOARD_HAS_PSRAM 
                 -mfix-esp32-psram-cache-issue 
                 -mfix-esp32-psram-cache-strategy=memw 
                 -DCORE_DEBUG_LEVEL=0
```

zekageri, 2023-02-04:
There is no specified dependency

ivankravets, 2023-02-05:
you need to remove "dependencies" from the `library.json` for each installed library into the `lib` folder. See the source of issue https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/library.json#L19

closed 2023-02-05 by ivankravets