platformio/platformio-core #4546: Custom program name causes error when running native tests

j4m3s, 2023-02-07:
Having followed the instructions [here](https://docs.platformio.org/en/latest/scripting/examples/custom_program_name.html) I now have a version number appended to my firmware binaries.  However, I now also get 
```
Building...
Testing...
[Errno 2] No such file or directory: '.pio/build/native/program'
```
when I try to run my native tests.  

Commenting the extra_script out in platformio.ini allows the tests to run fine again in the `native` .  The tests work ok in the embedded environment - the extra_script only causes the problem running native tests.

ivankravets, 2023-02-07:
Yes, the problem is here https://github.com/platformio/platformio-core/blob/develop/platformio/test/runners/readers/program.py#L78

We will think about how to fix it.

j4m3s, 2023-02-07:
That's really interesting - why would appending something to PROGNAME cause an issue there?  The updated PROGNAME doesn't seem to be referenced in that function, and the error message doesn't contain anything pointing to the appended PROGNAME either.  

(I should say... my python knowledge is minimal, so don't spend time replying if it's going to be obvious to others)

ivankravets, 2023-04-15:
Thanks for the report. Please re-test with `pio upgrade --dev`.

Does it work now?

closed 2023-04-15 by ivankravets

j4m3s, 2023-04-16:
Yes!  Brilliant, thank you :)

How do I get off the dev branch and back onto stable when the next version of platformio is released?

ivankravets, 2023-04-17:
See https://docs.platformio.org/en/latest/core/installation/development-version.html

commit 0d9ee75b (2023-04-15): Resolved an issue where native tests would fail if a custom program name was specified // Resolve #4546