platformio/platformio-core #4553: Debugger does not display value in native unit testing project

dtila, 2023-02-14:
Given a small PlatformIO project referencing ArduinoMock, the debugger does not display the value given by the mock.

[image]

The bug can be reproduced using this project: https://github.com/dtila/platfornio_unittesting

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 10

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.6

#### Steps to Reproduce

1. Checkout the project was given the link above
2. Hit a breakpoint in the unit test

You should see debugger displayed value is not the one returned by the mock.

However the mock works correctly, since running the tests, they pass successfully:
`Building...
Testing...
test\test_native_sut\SUTTests.cpp:54: TestMock  [PASSED]`

ivankravets, 2023-04-15:
@valeros , could you reproduce this issue? I can't make my `gdb` work on the latest macOS due to the system error `warning: unhandled dyld version (17)`.

valeros, 2023-04-16:
Strange, but I cannot reproduce:

[image]

ivankravets, 2023-04-17:
Thanks, @valeros!

closed 2023-04-17 by ivankravets