platformio/platformio-core #4555: Optimisation: specify directories to ignore during installation in library.json

RobMeades, 2023-02-16:
**Feature Request**

When developing a library we very likely store that library in Git but use it locally, so the `lib_deps` will be to the local disk; a customer may do the same if they have cloned the library locally, instead of giving a URL to `lib_deps`.

This is fine except that the library installation process ends up bringing with it the stuff from the `.git` directory, as well as the "wanted" files, leading to a much large copy than is necessary; for instance the `.git` directory in our library is 98 Mbytes in size, versus the wanted files which are half that, hence three times as much data is being copied than required.  On my Windows machine (which admittedly has work-mandated virus and anti-malware applications running on it) library installation can take many minutes each time, which is painful.

It would be a useful optimisation to be able to specify in `library.json` a [set of] directorie(s) to be excluded during the installation process.

RobMeades, 2023-02-18:
It would be good if this could be recursive as 90% of the 49 Mbytes of stuff I refer to as "wanted" above is actually another `.git` folder a bit deeper-down, in a Git sub-module.

Is there some form of "script" hook that might do what I want already?

ivankravets, 2023-03-28:
A few solutions:

1. Instead of reinstalling the local library o each build, please use [local folder symlinks](https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#local-folder)
2. Later when you are ready to publish the library to the registry, you can control "what is going to the user" using the [export](https://docs.platformio.org/en/latest/manifests/library-json/fields/export/index.html) field from the `library.json` manifest.

Did it help?

closed 2023-03-28 by ivankravets