platformio/platformio-core #4556: Serial Monitor filters does not work with "solution" for colorized output

BOBAH1248, 2023-02-17:
Many people love this wonderful tool, but sometimes they want more, especially since it is at least according to the documentation.

I found that it is based on `miniterm` part of `serial.tools`.
And then found that "solution" for `colorize` just write text as binary data:
https://github.com/pyserial/pyserial/blob/master/serial/tools/miniterm.py#L511-L517
So, colorization of output is done by OS terminal as expected.

Everything would be fine, but this solution also discards all filters! Include the best one `esp32_exception_decoder`.

Initially I faced this issue in PlatformIO IDE for VSCode under ESP32.
But research show that issue in `platformio.exe device monitor`, and in dependent 3th party submodule `serial.tools.miniterm`.

### Configuration

**Operating system**:
| OS Name	| Microsoft Windows 11 Pro |
| Version	| 10.0.22621 Build 22621 |

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.6

### Description of problem
The Serial Monitor filters are not work with "colorize solution":
```
monitor_filters = esp32_exception_decoder, direct, colorize
monitor_raw = yes   ; this is "colorize solution"
```

#### Steps to Reproduce

1. Use ESP32/ESP8266 solution with enabled colorized output to uart, with throwing an "Guru meditation error" (like "assert", "abort" and so)
2. When used `platformio.ini` with rows as in description, the colors works as expected, but `Backtrace:` from output is not parsed
3. If comment `monitor_raw` in `platformio.ini`, the colors don't works, but `Backtrace:` works as expected

### Actual Results
2. When used `platformio.ini` with rows as in description, the colors works as expected, but `Backtrace:` from output is not parsed

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

BOBAH1248, 2023-02-20:
It looks like, I managed to found way to avoid this issue on system, that use console with support of ANSI colors (ps: the VS Code support it) [refs pull request #1039](https://github.com/platformio/platform-espressif32/pull/1039)

ivankravets, 2023-03-14:
What is your goal? You want to use filters but you disabled them with `monitor_raw`. See docs https://docs.platformio.org/en/latest/projectconf/sections/env/options/monitor/monitor_raw.html

The `esp32_exception_decoder` filter already sends ANSI codes (colors), you don't need extra `colorize`.

Just try this:

```
monitor_filters = direct, esp32_exception_decoder
```

BOBAH1248, 2023-03-14:
> What is your goal?

My goal is make it usable (in the most possible cases) and clarify the documentation.

> You want to use filters but you disabled them with `monitor_raw`.

https://github.com/platformio/platformio-vscode-ide/issues/2750#issuecomment-956150843

ivankravets, 2023-03-14:
The `colorize` filter does "Apply different colors for received and echo". It just colorizes input/output with different colors. Should we improve our docs? We would be thankful if you point us to the place where is not clear which filter you should to use.

BOBAH1248, 2023-03-14:
To be able to see what I sent and what I received from the device in different colors, this should in no way affect what the device sends me, additionally tinting its parts of the lines.
Moreover, all the logs from the device begin to tint the line at the very beginning of the line and cancel it at the very end of the line.
So it turns out that `colorize` should add the color character after the newline (<CR>/<LF>), and mark the color immediately before the next linefeed.
That is, it should "frame" the output from the device, which means it cannot technically affect the colors inside the line.

ivankravets, 2023-03-14:
How about if we just remove `colorize` from the docs? I see it adds more problems than helps developers. Please note that we can't fix the `colorize` filter because it is the part of PySerial package.

BOBAH1248, 2023-03-14:
Agreed, the `colorize` (as well some another like `direct`) is part of `miniterm` and cannot be easily changed. And due to it adds problems it will be good to remove from docs or marked as "use at your own risk"

ivankravets, 2023-03-16:
Resolved in https://github.com/platformio/platformio-docs/commit/67c016f8b7f79d2ca5c8cb8361887da8ff4fb240

old-Jerry, 2026-01-08:
I found that using Microsoft's Serial Monitor plugin in VS Code displays the espidf terminal window in color very well; it's standard. However, I think it would be great if platformio-core could support calling this serial terminal and automatically releasing it during download and burning.
What we love most about Platform.io are its automatic serial port recognition, automatic opening of the serial port after programming, and automatic closing of the serial port during programming.