platformio/platformio-core #4567: After export Arduino IDE project, i have errors: `.dram0.bss' will not fit in region `dram0_0_seg' \ DRAM segment data does not fit \ region `dram0_0_seg' overflowed by 488 bytes

Voha888, 2023-03-11:
This project https://github.com/Circuit-Digest/ESP32-Oscilloscope/tree/main/ESP32_Oscilloscope
, i converted to PIO format for three ways:

1. Classic C++: created file structure with .cpp and .h files
2. All .ino files cobine in one big .cpp file
3. Using PIO option "Import Arduino Project".

In all three ways, i have the same error:

```
Linking .pio\build\esp32dev\firmware.elf
c:/users/user/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
c:/users/user/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/user/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/user/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 488 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1

```

But in Arduino IDE its project compiling and working well. For ESP32 versions 2.xx need the little change. The line:

`i2s_read_bytes(I2S_NUM_0, (char*)&i2s_buff[i * NUM_SAMPLES],  NUM_SAMPLES * sizeof(uint16_t), portMAX_DELAY);`

change to:

```
size_t bytes = 0;    
   i2s_read(I2S_NUM_0, (char*)&i2s_buff[i * NUM_SAMPLES],  NUM_SAMPLES * sizeof(uint16_t), &bytes, portMAX_DELAY); 
```

maxgerhardt, 2023-03-13:
Solved per [community post](https://community.platformio.org/t/after-export-arduino-ide-project-i-have-errors-dram0-bss-will-not-fit-in-region-dram0-0-seg-dram-segment-data-does-not-fit-region-dram0-0-seg-overflowed-by-488-byte/32529).

ivankravets, 2023-03-14:
@maxgerhardt , thanks for the provided help!

closed 2023-03-14 by ivankravets