platformio/platformio-core #4571: Upgrade cppcheck to 2.10

grisharav, 2023-03-15:
What kind of issue is this?

- [X ] **PlatformIO Core**.
     Would like to use latest version of cppcheck, 2.10. The latest available from the registry https://registry.platformio.org/tools/platformio/tool-cppcheck is ~9 months old

### Configuration

**Operating system**: Windows x64

**PlatformIO Version** (`platformio --version`):
6.1.6

### Description of problem
New cppcheck tool properly identifies several best practices (specifically for my case regarding `string_view`) and reduces false warnings

#### Steps to Reproduce

Pass `string_view` by value and you will see cppcheck issuing a warning about suggestion to pass it by reference.

grisharav, 2023-03-15:
Note that if I manually replace the binary in the `.platformio\packages\tool-cppcheck\` folder, the analysis works as expected

valeros, 2023-04-25:
Hi @grisharav, thanks for pinging us. We are aware that there were updates of the Cppcheck tool. Currently, we're evaluating how to properly integrate the new versions of Cppcheck as some of the advanced functionality is now moved to Cppcheck Premium package (e.g. MISRA checks). It may be harmful if we update this package silently as some already configured PlatformIO projects expect this functionality.

red-scorp, 2023-08-01:
Check All command fails miserably with most of Arduino-compatible toolchains because of outdated cppcheck package. I can confirm, replacing tool-cppcheck files with newer versions repairs the Check All feature. Support of MISRA is a good feature, but not essential for most of developers. ;-)

BTW, cppcheck 2.11 is out.

closed 2023-08-17 by valeros

red-scorp, 2023-08-17:
@valeros thank man! 💪😄

red-scorp, 2025-07-04:
```
C:\Users\XXX\.platformio\packages\tool-cppcheck>cppcheck.exe --version
Cppcheck 2.11
```
```
C:\Program Files\Cppcheck>cppcheck.exe --version
Cppcheck 2.17.1
```

Do we need to create new issue for each cppcheck SW update?
Please update the package again!