platformio/platformio-core #4575: Wrong repository included in .pio/libdeps

Bascy, 2023-03-18:
What kind of issue is this?

- [x] **PlatformIO Core**.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 11

**PlatformIO Version** (`platformio --version`): 6.1.6

### Description of problem
When adding the repo `https://github.com/jbeder/yaml-cpp` to the lib_deps property in `platformio.ini` it will only load `googletest repository in `.pio/libdeps`.

The same happend after I made a clone of the yaml-cpp repo to my own environment. After i removed the `test` folder the problem went away

#### Steps to Reproduce

1. Add ``https://github.com/jbeder/yaml-cpp` to libdeps
2. Build
3.

### Actual Results
`.pio/libdeps` contains clone of googletest repo

### Expected Results
`pio/libdeps` contains clones of both yaml-cpp and googetest repos

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
default_envs = debug 

[env]
platform = espressif32
board = esp32dev
board_build.partitions = min_spiffs.csv
framework = arduino

lib_ldf_mode = chain
lib_deps = 
    https://github.com/jbeder/yaml-cpp

build_flags = 
  -fmerge-constants
  -Wfatal-errors

[env:debug]
build_type = debug 

### Additional info

ivankravets, 2023-03-18:
The https://github.com/jbeder/yaml-cpp is not a valid PlatformIO package. It is not clear where is the root of the source. Please fork that library and add https://docs.platformio.org/en/latest/manifests/library-json/index.html manifest.

Example https://github.com/espressif/esp32-camera/blob/master/library.json

closed 2023-03-18 by ivankravets

Bascy, 2023-03-18:
ok, I agree that its not a platformIO package .. but why doesn't the package itsself get cloned?

ivankravets, 2023-03-18:
The PlatformIO looks recursively for the manifest, and it seems it founds this https://github.com/jbeder/yaml-cpp/tree/master/test/gtest-1.11.0