platformio/platformio-core #4576: escaping $ in build_flags

jschoch, 2023-03-18:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: windwos

**PlatformIO Version** (`platformio --version`):

### Description of problem

using env.StringifyMacro() doesn't escape special characters
also it isn't obvious what the escapes are, for instance "$$" to escape "$" isn't obvious.  "\\$" produces "/$" which is also baffeling.  The docs only say to escape characters they don't say how you do it.

Request: document escape characters and special characters used in .ini build_flags and env variables

ivankravets, 2023-03-18:
Could you provide a `platformio.ini` example?

ivankravets, 2023-04-12:
@bdbaddog, @mwichmann, sorry for disturbing you directly.

Could you share with us any ideas on how to disable SCons variable substitution for the CPPDEFINE/Macro?

The `env.Append(CPPDEFINES=[("WIFI_PWD", "123$$a$$")])` produces `... -DWIFI_PWD=123$ ... ` in a verbose mode where SCons tries to evaluate `$xxxx`.

Thanks in advance.

bdbaddog, 2023-04-12:
Two options:
1. create your own escape function and set via `env['ESCAPE']` (See manpage)
2. Change `env.Append(CPPDEFINES=[("WIFI_PWD", "123$$a$$")])` 
 to `env.Append(CPPDEFINES=[("WIFI_PWD", "123\$$\$$a\$$\$$")])`  
$$ substitutes to $, then the \ should escape it for the SHELL.

I'd say go ahead and add your example to this existing github issue (if you agree that it's basically the same issue?)
https://github.com/SCons/scons/issues/1093

ivankravets, 2023-04-12:
@bdbaddog, thanks for the prompt help!

The 2nd option will work well for us. I tried it and it still "eats" char `a`. The `env.Append(CPPDEFINES=[("WIFI_PWD", "123\$$\$$a\$$\$$")])` produces ` -DWIFI_PWD=123\$\\$\$`.

ivankravets, 2023-04-12:
@mwichmann , I tried it with `("WIFI_PWD", env.Literal('\\\\$$ORIGIN'))` and it produces `-DWIFI_PWD=\\`.