platformio/platformio-core #4578: Build flags -include ... does not update dependency tree

mcspr, 2023-03-20:
### Configuration

**Operating system**:

Any

**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 6.1.7a3

### Description of problem

`-include foo.h`, `-include bar.h` specified in `build_flags` / `build_src_flags` does not create a direct dependency on the resulting program like `#include` in the source does

#### Steps to Reproduce

Given the project structure

```console
> tree
.
├── platformio.ini
└── src
    ├── main.cpp
    └── vendor
        └── custom.h

3 directories, 3 files
```

platformio\.ini
```ini
[env:native]
platform = native
build_src_flags = -include src/vendor/custom.h
```

src/main.cpp
```cpp
#ifndef FOO
#warning "default value"
#endif

int main() {
}
```

src/vendor/custom.h
```cpp
#define FOO
```

(files duped at https://gist.github.com/mcspr/753a18cc3f0a3f4f907a79a7afd8e4ea)

```console
> pio run -s -t clean

> pio run
Processing native (platform: native)
------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/native/src/main.o
Linking .pio/build/native/program
=========================== [SUCCESS] Took 0.69 seconds ===========================
```

### Actual Results

Modifying `src/vendor/custom.h` does not rebuild the `.pio/build/native/program`
```console
> echo "#define BAR" > src/vendor/custom.h
m.h

> pio run
Processing native (platform: native)
------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
`.pio/build/native/program' is up to date.
=========================== [SUCCESS] Took 0.57 seconds ===========================
```

Since no dependency was created for the project file
```console
> env SCONSFLAGS="--tree=all" pio run
Processing native (platform: native)
------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
`.pio/build/native/program' is up to date.
+-.pio/build/native/program
  +-.pio/build/native/src/main.o
  | +-src/main.cpp
  | +-/usr/bin/g++
  +-/usr/bin/g++
```

### Expected Results

Program should rebuild when include file is modified

### Additional info

https://github.com/SCons/scons/issues/3064
Upstream seems to have this on hold because of PCH (aka precompiled headers) support, but idk if it is relevant here

At last one instance of `-include` in PIO documentation
https://docs.platformio.org/en/stable/platforms/espressif8266.html
(and `umm_malloc_cfg.h` happens to be tracked from a different file anyway, so dependency is added through other means)

mcspr, 2023-03-31:
Nope, this meant to include a specific file. Relative path meant to originate from `$PROJECT_DIR` i.e. where `platformio.ini` is
https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html#index-include

DaveDave9994, 2023-04-20:
Im having almost the same issues.
 ihave tried serval different include paths in platforio.ini  but none seem to work.
`build_flags =
	I-include
	I-C:\Users\bucko\OneDrive\aElectronics\Projects\BuckosLearning-PH7\include
	I-C:\Users\bucko\OneDrive\aElectronics\Projects\BuckosLearning-PH7\include\blProjConfig_PH7.h`
pio run -v
` pio run -v
Processing portenta_h7_m7 (platform: ststm32; board: portenta_h7_m7; framework: arduino; lib_deps: arduino-libraries/ArduinoBLE@^1.3.4, fmalpartida/LiquidCrystal @ ^1.5.0, https://github.com/maxgerhardt/Arduino_PortentaBreakout/archive/refs/heads/main.zip; build_flags: I-include, I-C:\Users\bucko\OneDrive\aElectronics\Projects\BuckosLearning-PH7\include, I-C:\Users\bucko\OneDrive\aElectronics\Projects\BuckosLearning-PH7\include\blProjConfig_PH7.h)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/portenta_h7_m7.html
PLATFORM: ST STM32 (15.6.0) > Arduino Portenta H7 (M7 core)
HARDWARE: STM32H747XIH6 480MHz, 511.35KB RAM, 768KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduino-mbed @ 4.0.2 
 - tool-dfuutil @ 1.11.0 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 46 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoBLE @ 1.3.4 (License: Unknown, Path: C:\Users\bucko\OneDrive\aElectronics\Projects\BuckosLearning-PH7\.pio\libdeps\portenta_h7_m7\ArduinoBLE)
|-- LiquidCrystal @ 1.5.0 (License: Unknown, Path: C:\Users\bucko\OneDrive\aElectronics\Projects\BuckosLearning-PH7\.pio\libdeps\portenta_h7_m7\LiquidCrystal)   
|   |-- Wire (License: Unknown, Path: C:\Users\bucko\.platformio\packages\framework-arduino-mbed\libraries\Wire)
|-- Arduino_PortentaBreakout @ 1.0.1 (License: Unknown, URI: https://github.com/maxgerhardt/Arduino_PortentaBreakout/archive/refs/heads/main.zip, Path: C:\Users\bucko\OneDrive\aElectronics\Projects\BuckosLearning-PH7\.pio\libdeps\portenta_h7_m7\Arduino_PortentaBreakout)
|   |-- Ethernet @ 1.0.0 (License: Unknown, Path: C:\Users\bucko\.platformio\packages\framework-arduino-mbed\libraries\Ethernet)
|   |   |-- SocketWrapper @ 1.0 (License: Unknown, Path: C:\Users\bucko\.platformio\packages\framework-arduino-mbed\libraries\SocketWrapper)
|   |-- PDM @ 1.0 (License: Unknown, Path: C:\Users\bucko\.platformio\packages\framework-arduino-mbed\libraries\PDM)
|   |-- SPI (License: Unknown, Path: C:\Users\bucko\.platformio\packages\framework-arduino-mbed\libraries\SPI)
|   |-- Wire (License: Unknown, Path: C:\Users\bucko\.platformio\packages\framework-arduino-mbed\libraries\Wire)
Building in release mode`