platformio/platformio-core #4582: Marlin 2.1.2 Compile Issue SKR 1.4

BinLagging, 2023-03-26:
Hi,
I have tried everything to get past this issue. I tried several PCs and Laptops with different builds of windows but no luck. My compile of firmware keeps failing. Any ideas what am I missing? Below is my terminal. I am using Platformio Core 6.1.6 and Home 3.4.3

Library Manager: U8glib-HAL@0.5.2 has been installed!
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Servo @ 1.0.0
|-- TMCStepper @ 0.7.3
|-- Adafruit NeoPixel @ 1.5.0
|-- U8glib-HAL @ 0.5.2
Building in release mode
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MinSerial.cpp.o
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_flash.cpp.o
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_wired.cpp.o
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\fast_pwm.cpp.o
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
The system cannot find the path specified.
The system cannot find the path specified.
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\include\i2c_util.c.o
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
arm-none-eabi-gcc: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
arm-none-eabi-gcc: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\main.cpp.o
The system cannot find the path specified.
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\MinSerial.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\fast_pwm.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\include\i2c_util.c.o] Error 1
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\main.cpp.o] Error 1
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
The system cannot find the path specified.
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\tft\tft_spi.cpp.o] Error 1
========================================= [FAILED] Took 16.53 seconds =========================================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:00:16.527
==================================== 1 failed, 0 succeeded in 00:00:16.527 ====================================

ivankravets, 2023-03-28:
See https://community.platformio.org/t/cant-build-my-project/17312

closed 2023-03-28 by ivankravets

BinLagging, 2023-03-28:
> See https://community.platformio.org/t/cant-build-my-project/17312

Thank you that worked!!!!