platformio/platformio-core #4585: New Project Wizard does not produce platformio.ini and other subfolders

PeterDollar, 2023-03-31:
- [ ] **PlatformIO Core**.
When attempting to create an new project via the Project Wizard, the creation of some folders and platformio.ini fails.  The only file or folder created under the root is src.  And it contains one file, main.cpp.

This means that there is no way to compile or build...

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`):  Core 6.1.6, Home 3.4.3

### Description of problem
When attempting to create an new project via the Project Wizard, the creation of some folders and platformio.ini fails.  The only file or folder created under the root is src.  And it contains one file, main.cpp.

#### Steps to Reproduce

1. Start vscode
2. Select platformio
3. Create new project
4.  Select new project button (start new project wizard)
5.  Specify project name, board (UNO), framework and location 

### Actual Results
 Project wizard does not complete properly and only creates src folder and main.cpp

### Expected Results
The root folder should contain the following folders/files
.pio
.vscode
include
lib
src
test
.gitignore
platformio.ini

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here... DOESN'T EXIST !
```

**Source file to reproduce issue:**
```cpp
Insert here...  IS INDEPENDENT OF SOURCE CODE
```

### Additional info
Have exhausted all inputs from the community.  See [Why is no platformio.ini file being created by new project wizard?](https://community.platformio.org/t/why-is-no-platformio-ini-file-being-created-by-new-project-wizard/32923)

PeterDollar, 2023-04-14:
Issue resolved

closed 2023-04-14 by PeterDollar

ivankravets, 2023-04-14:
How did you solve this issue?

PeterDollar, 2023-04-14:
That's the issue I have Ivan.  I don't know.

All I can suggest is that I installed ESP-IDF via its Windows installer and maybe that changed something? It's the only thing i changed on my PC.

I did notice that once I started using VSCode again, there was a new ESP-IDF icon under the Platformio icon inside VScode.  I don't use it but it's there.

Unfortunately I can't say that I know what fixed the issue as I had given up trying to fix it,  I was just looking for other ways to continue coding.

Peter