platformio/platformio-core #4590: Newest toolchain-gccarmnoneeabi 1.110301.0 package broken

maxgerhardt, 2023-04-04:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Win 10 x64

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.7a2

### Description of problem

When attempting to use the latest 11.3.1 version of the arm-none-eabi GCC/G++ toolchain as uploaded in https://registry.platformio.org/tools/platformio/toolchain-gccarmnoneeabi/versions, compilation of most firmwares immediately break because the file `bits/c++config.h` does not exist in the toolchain.

#### Steps to Reproduce

1. Create a new Bluepill F103C8 + Arduino project
2. Add empty sketch code
3. Hit compile --> it should compile normally
4. Add `platform_packages = platformio/toolchain-gccarmnoneeabi@1.110301.0` to the `platformio.ini`
5. Build again

### Actual Results

```
Compiling .pio\build\bluepill_f103c8\FrameworkArduinoVariant\variant_PILL_F103Cx.cpp.o
In file included from c:\users\max\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\stdlib.h:36,
                 from C:\Users\Max\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:21,
                 from C:\Users\Max\.platformio\packages\framework-arduinoststm32\variants\STM32F1xx\F103C8T_F103CB(T-U)\variant_PILL_F103Cx.cpp:15:
c:\users\max\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```
### Expected Results

Successfull compilation.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
platform_packages =
   ; doesn't work
   platformio/toolchain-gccarmnoneeabi@1.110301.0
   ; works
   ;platformio/toolchain-gccarmnoneeabi@1.100301.220327
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() {}
void loop() {}
```

### Additional info

Affects Marlin compilation for NXP LPC176x chips per https://community.platformio.org/t/compile-errors-c-11-3-1-bits-c-config-h10-fatal-error/33038/6?u=maxgerhardt

valeros, 2023-04-05:
Hmm, the `11.3.1` toolchain was uploaded for the Teensy dev-platform. It seems that packages were specifically compiled for the Teensyduino's needs because there are no issues with this toolchain when projects are compiled for Teensy boards. Apparently, that packages cannot be used outside of the Teensy dev-platform.

@ivankravets Does it make sense to reupload that toolchain packages with Teensy-specific names to avoid confusion?

maxgerhardt, 2023-04-05:
Very intersting. So the above error occurs when even the simplest most 
```cpp
#include <cstdlib>
```
is done. **But** in Teensy project, the file is actually found at
```
C:\Users\<user>\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\arm-none-eabi\thumb\v7e-m+dp\hard\bits\c++config.h
```

When comparing the toolchains against each other, the new toolchain seems to be lacking a few of these architecture folders

[image]

closed 2023-04-05 by valeros

p3p, 2023-04-05:
will the incomplete `toolchain-gccarmnoneeabi @ 1.110301.0 (11.3.1)` be removed from the registry?

valeros, 2023-04-05:
@p3p Yes, it will be removed.