platformio/platformio-core #4592: lib_deps download different repository

dibalavs, 2023-04-08:
- [ ] **Development Platform or Board**.
heltec_wifi_lora_32_V2

- [ ] **PlatformIO Core**.
PS C:\Work\repos\KaRadio32_4> pio --version
PlatformIO Core, version 6.1.6

### Configuration

**Operating system**:
Windows 10 22h2

**PlatformIO Version** (`platformio --version`):
PS C:\Work\repos\KaRadio32_4> platformio --version
PlatformIO Core, version 6.1.6

### Description of problem
when I type:
lib_deps = 
    https://github.com/espressif/esp-protocols.git

in platformio.ini - it downloads completely different repository.

#### Steps to Reproduce

1. add lib_deps in platformio.ini
2. rebuild project
3. see at .pio/libdeps/heltec_wifi_lora_32_v2/*

### Actual Results
[image]

### Expected Results
[image]

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; pio run -t menuconfig
[platformio]
src_dir = main

[env:heltec_wifi_lora_32_V2]
platform = espressif32
board = heltec_wifi_lora_32_V2
framework = espidf
monitor_speed = 115200
monitor_flags=
	--echo
	--raw
board_build.partitions = partitions.csv

;-Wno-int-conversion -Wno-incompatible-pointer-types 
build_flags = -Wno-format
build_unflags = -Werror=format -Werror=all -Werror=int-conversion

lib_deps = 
    https://github.com/espressif/esp-protocols.git

platform_packages =
    platformio/framework-espidf@^3.50001.0                                                                                                        
    toolchain-xtensa-esp32 @ 11.2.0+2022r1     

board_upload.maximum_size = 16777216
board_upload.flash_size = 16MB

; To enable ccache speedup, install ccache to be visible in PATH
; https://github.com/ccache/ccache/releases
board_build.cmake_extra_args = -DCCACHE_ENABLE=ON
extra_scripts =
    ;pre:use_ccache.py
    post:use_ccache.py
#build_flags = -v
```

maxgerhardt, 2023-04-08:
1. An ESP-IDF component cannot be added the project by using `lib_deps`. You must either download it into the `components/` folder of your project or add it as git submodule. See [documentation](https://docs.platformio.org/en/stable/frameworks/espidf.html) and [reference project](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-aws-iot).
2. Because PlatformIO tries to interpret the `https://github.com/espressif/esp-protocols.git` repo as a library, but there is no `library.properties` or `library.json` at the root, it starts to traverse the folders until it finds something that looks like a library. And since `common_components/protocol_examples_common` does have an `include/` folder it thinks that's the actual library and uses only that subtree.

ivankravets, 2023-04-08:
@maxgerhardt , thanks for the help! 🙏

closed 2023-04-08 by ivankravets