platformio/platformio-core #4594: latest dev does break compile of any Tasmota Env variant

Jason2866, 2023-04-11:
@ivankravets  Only change is the pio core. Stable pio does work 
Failing GH Actions of the Tasmota specials env
https://github.com/Jason2866/Tasmota-build/actions/runs/4670109313

ivankravets, 2023-04-12:
@mcspr, could it be caused by your PR https://github.com/platformio/platformio-core/pull/4586 ? Should we revert it?

ajlennon, 2023-04-12:
Hi all - Staars flagged this to me on the Tasmota Discord and I can confirm that if I revert that line of regexp the build works here.

ivankravets, 2023-04-12:
Thanks for the confirmation. The changes were reverted https://github.com/platformio/platformio-core/commit/58a1d5d96e6a1259dcb5d12b7b103eff4bbdd1fd

Please re-test the latest development version.

closed 2023-04-12 by ivankravets

ajlennon, 2023-04-12:
Thanks @ivankravets - this all seems to be building OK now with `PlatformIO has been successfully upgraded to 6.1.7a4` which I assume is the reversion

https://github.com/DynamicDevices/Tasmota/actions/runs/4678076096

ajlennon, 2023-04-12:
No worries at all - I'll leave my Tasmota fork CI running your latest development version if it helps :)

mcspr, 2023-04-13:
> @mcspr, could it be caused by your PR #4586 ? Should we revert it?

[Library include in driver file](https://github.com/arendst/Tasmota/blob/870874d181596ab3a62abeb91ec2a2f0122e2c6a/tasmota/tasmota_xdrv_driver/xdrv_05_irremote_full.ino#L62) is unknown at the top of the resulting .ino.cpp. For PR to work, Tasmota would have to move `#include <IRremoteESP8266.h>` to the top of tasmota.ino

*Not* generating prototypes still feels like a bug :/
(btw I forgot about return types, `:` should be matched there too)

Comparing to arduino-cli, it also does not like `#include` in extra .ino files.
But, same as PIO, it would respect existing includes in the main file and inject declarations just after
```console
~/sketch> cat sketch.ino
#include <IRremoteESP8266.h> // <--- everything breaks if this line is removed

void setup() {
}

void loop() {
}
```
```console
~/sketch> cat extra.ino
#include <IRremoteESP8266.h>

void foo(const stdAc::state&) {
    puts("bar");
}
```
```console
~/sketch> arduino-cli compile -b esp8266com:esp8266:generic --preprocess
```
```cpp
#include <Arduino.h>
#line 1 "/home/runner/sketch/sketch.ino"
#include <IRremoteESP8266.h>

#line 3 "/home/runner/sketch/sketch.ino"
void setup();
#line 6 "/home/runner/sketch/sketch.ino"
void loop();
#line 3 "/home/runner/sketch/extra.ino"
void foo(const stdAc::state&);
#line 3 "/home/runner/sketch/sketch.ino"
void setup() {
}

void loop() {
}

#line 1 "/home/runner/sketch/extra.ino"
#include <IRremoteESP8266.h>

void foo(const stdAc::state&) {
    puts("bar");
}
```