platformio/platformio-core #4596: Log2file - Store log files in user specific folder

JeroenIoT, 2023-04-12:
Hello,

I am the feature log2file in platformio. This is a very useful feature which I do use a lot.

- [ ] **Feature Request**.
- The problem I am facing is that the number of log files pile up in my project's root folder.
Often there are too many and I have to move the log files to another folder.

- My request is that there is another variable in platformio.ini which allows me to specify the folder (within the project folder) in which the logs files are stored.
For example:

      Platformio.ini:
         monitor_filters = default, time, log2file ; 
         monitor_log_dir = logs;  (new / not exisiting at this moment)

      So the new variable "monitor_log_dir = logs" would place the logs in 
      C:\Users\username\Documents\PlatformIO\Projects\project_root\logs\
- If the folder does not exist it should be created.

Hope this is clear and feedback is welcome. Also discussed this topic [here](https://community.platformio.org/t/log2file-how-can-log-files-be-stored-in-user-specific-folder/33081)

ivankravets, 2023-04-12:
Thanks for the report. PlatformIO will use the `libs` folder for saving monitor logs. 

Please re-test with `pio upgrade --dev`.

closed 2023-04-12 by ivankravets

JeroenIoT, 2023-04-13:
- Upgraded to PlatformIO Core, version 6.1.7a4
- **Tested**:
- Compiled with no changes to platformio.ini
      `monitor_filters = default, time, log2file   ; adds time tracking to the TERMINAL and enables logging`
- Log files are stored in project folder in folder `logs` (_not_ in `libs `as you wrote above)
- Example: 
      `C:\Users\username\Documents\PlatformIO\Projects\pioProjectName\logs`

commit 5c3ae15b (2023-04-12): Store device monitor logs in the project "logs" directory // Resolve #4596