platformio/platformio-core #4599: Unable to build projects with a custom fork of espidf

shlomozippel, 2023-04-13:
- [X] **PlatformIO Core**.

### Description of problem

Can't build an espressif32 project using a custom fork of espidf. The bug is in the platformio core `version.py:get_original_version` which doesn't have version names with a `+sha.gitsha` in them, something that happens when using a custom espidf fork.

------------------------------------------------------------------

### Configuration

**Operating system**: OSX

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.6

#### Steps to Reproduce

1. Fork espidf repo, add library.json
2. Build a project with the custom version of espidf
```
[env]
platform = espressif32
framework = espidf, arduino
platform_packages = framework-espidf @ https://github.com/chroma-tech/esp-idf#ethernet_fix
```

### Actual Results

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.1.0+sha.41f7000) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.6+sha.e1a9a602 
 - framework-espidf @ 3.40404.0+sha.2382fa6 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
TypeError: can only concatenate str (not "NoneType") to str:
  File "/Users/shlomo/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 187:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/shlomo/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/shlomo/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/shlomo/.platformio/platforms/espressif32/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/Users/shlomo/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/Users/shlomo/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/piobuild.py", line 60:
    env.ProcessProgramDeps()
  File "/Users/shlomo/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/Users/shlomo/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/piobuild.py", line 120:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/Users/shlomo/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/Users/shlomo/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/piobuild.py", line 331:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/Users/shlomo/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 660:
    return method(*args, **kw)
  File "/Users/shlomo/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/shlomo/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/shlomo/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1191:
    install_python_deps()
  File "/Users/shlomo/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1098:
    python_exe_path = get_python_exe()
  File "/Users/shlomo/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1169:
    env.subst("$PROJECT_CORE_DIR"), "penv", ".espidf-" + idf_version)
```

### Expected Results

The build should work

### Additional info

Patched platformio core with a fix which worked for me: https://github.com/platformio/platformio-core/pull/4598

ivankravets, 2023-04-15:
Custo ESP-IDF forks are not supported.

closed 2023-04-15 by ivankravets

shlomozippel, 2023-04-15:
If anyone encounters this in the future and wants a workaround without fixing the issue in pio core, I was able to add this custom pio pre:script:
```
Import("env")
platform = env.PioPlatform()
pkg = platform.get_package("framework-espidf")
pkg.metadata.version.build = None
```

quantumboar, 2024-02-04:
shlomozippel's answer above almost worked for me. Apparently, on newer platform versions the function get_original_version() checks for actual versions that, for some reason, must have a field at position 1 that's larger than 99. If the solution above doesn't work for you, you might want to try:

`platform = env.PioPlatform()
pkg = platform.get_package("framework-espidf")
pkg.metadata.version = "4.100.100"`

(adjust the version to something suitable for you, just remember the 2nd field has to be > 99)